MockbaTheBorg/RunCPM #116: Games on RunCPM

dadecoza, 2020-08-28:
Hi!

Great project. :)

I'm running the posix build and games like catchum and ladder runs way too fast.

Is there a way to slow it down?

Regards,
Johannes

MockbaTheBorg, 2020-08-28:
Hi Johannes,

The only way I can think of would be adding a delay loop onto the CPU code.
This would be on CPU.h.

I will think about it and get back to you.
Maybe it is just a simple do-nothing FOR loop.

Cheers,
Marcelo.

MockbaTheBorg, 2020-08-29:
Hi Johannes,

I have been thinking about this "slowdown" simulation and I believe that the best way to do it would be to insert a delay at like 1455 of cpu.c, which is where every instruction starts to be executed.
A 4Mhz Z80 has a .25uSec clock period (T-Cycle), and each instruction cycle (C-Cycle) takes from 4 to 23 times that, so you might need to lose at least 4 times this amount of time to get realistic results.
It will be averaged, because each Z80 instruction takes a different number of clock cycles to execute, but you can bring the number up until you are satisfied with the game's speed.

Keep in mind that a "do nothing" FOR loop might end up optimized out by the compiler, and thus yield no effect. Take that in consideration.

This is what came to mind. Anything other than this would me a more complicated solution, like tracking from a table the number of clock cycles an instruction takes, and then keep track of current time and previous time, so you only start the execution of the next instruction when the previous instruction should have finished. 
This would be more precise, so a program that executes DEC (HL) would indeed be two times faster than if it used DEC (ii+n) instead. But this type of timing should not matter for a video game, and we don't do any timing related things on RunCPM, so it would be overkill.

I recommend you give it a try and also join our Discord channel if you haven't already.

Cheers,
Marcelo.

closed 2020-08-29 by MockbaTheBorg

MockbaTheBorg, 2020-08-29:
[image]

I just tried this on Visual Studio. Seems to work fine, you will have to play with the number.

dadecoza, 2020-08-31:
@MockbaTheBorg thank you so much for the reply!

I did implement the "do nothing" loop but as you said, I think the compiler ignores it. I will play around a bit at least you have pointed me in the right direction.

I will also join the Discord channel.

You can close this issue.

Regards,
Johannes