MockbaTheBorg/RunCPM #118: Problems compiling RunCPM v4.4 with DJGPP

guidol70, 2020-10-01:
Today I did try to compile RunCPM v4.4 with DJGPP under Win98 (and Win7-32bit) but it seems that I have problems with the "make"-version.

First I did try the follwoing "packages"
djdev205.zip
faq230b.zip
bnu234b.zip
gcc930b.zip
gpp930b.zip
mak43b.zip
pdcur39a.zip

with mak43b.zip I do get a 
rmvfromfree: memory fouled
sigabrt
like on https://usenetarchives.com/view.php?id=comp.os.msdos.djgpp&g=22843&p=0

So I did try older mak's like mak41br2.zip or mak421b.zip, but with these version I do get (after some warnings) a
make.exe: Error 1   OR
make.exe: Error 2

Could you please tell which version-combination did you use (or a person who did successful did compile RunCPM with djgpp)

Thanks in advance ;)

MockbaTheBorg, 2020-10-01:
Hi @guidol70 ,

It has been a long long time since I used DJGPP, but I remember it used to compile with no issue.
The big problem is that I do not have the computer where I did it anymore.
I would suggest you also bring this question to our Discord group too, as there might be someone there who did it.

Cheers,
Marcelo.

MockbaTheBorg, 2020-10-02:
I think this is related to #110 and #112.
We made a whole lot of changes to _findfirst and _findnext on the other platforms.
This was done actually with help of others to bring more compatibility to the filesystem, however we left the DOS port behind.
My proposal would be: adapt these two functions on abstract.h to what is on abstraction_posix.h, however I do not have a DOS environment to do it anymore.
I am thinking about moving this issue to "Won't Fix" and start phasing out the DOS port. :(

guidol70, 2020-10-03:
I did wrote about my "successfull" compile-results at
(see the "Code"-part of this message in this thread)
https://forum.classic-computing.de/forum/index.php?thread/21563-runcpm-compile-via-djgpp-auf-dos-win9x-klappt-nicht/&postID=256644#post256644

The only problem now is that this RunCPM-binary doesnt show all files at a DIR-listing 
(using your Standard-A-Drive-Content)
So MBASIC.COM isnt listed as example, but could be startet from the CP/M-prompt :)

Maybe you got a small patch/solutiosn for the missing files at the DIR-listing?

[image]
[abstract_h_user_tux.zip](https://github.com/MockbaTheBorg/RunCPM/files/5322042/abstract_h_user_tux.zip)

MockbaTheBorg, 2020-12-21:
Support for MS-DOS/DJGPP officially removed. It's been fun running an emulator inside an emulator.
Farewell MS-DOS.

closed 2020-12-21 by MockbaTheBorg

guidol70, 2021-02-03:
@MockbaTheBorg 
Had today some findings:
I figured out how to start the working 32bit (TDM-GCC compiled) version of RunCPM v4.8 under DOS (7.1 Win98SE or 8.0 WIn Millenium)

you could use HX DOS DPMI32 Extender v2.17 (for pure DOS - not in graphical Windows)
(this extender also needs a copy of msvcrt.dll and crtdll.dll)
https://sourceforge.net/projects/hx-dos/files/
or
in graphical Windows 98SE you could use the KernelEx Core (also 32bit kernel extension)
https://msfn.org/board/topic/173233-kernelex-45-core-updates-45201617/
to start the 32bit TDM-GCC-32-compiled version in DOS

BUT also the on Win10 working binary which shows ALL files while DIR does show the same small amount of files for DIR while using a extender and DOSLFN. So it seems to be outside of RunCPM.

DOSLFN is the Long file Name support for DOS. Without loading DOSLFN before RunCPM does show "No File" for the DIR -command :(
http://adoxa.altervista.org/doslfn/index.html

[image]

"Strange" sideeffect FRACTAL.BAS on B/0 sint listed, but when copied as 12345678.BAS or FRACTAL1.BAS then the files would be listed BUT not in alphabetical order

When I copy 12345678.BAS from drive B:/0 to A:/0 PIP A:=B:12345678.BAS then this file is listed as the last file and FRACTAL1.BAS is listed AFTER 12345678.BAS

Also did try this with FreeDOS v1.3 and the extender HX DOS - same result.
So why does the 32Bit binary work fine under Win10 and not under the extender

MockbaTheBorg, 2021-02-04:
Hi @guidol70 ,

This is the problem with DOS. Over time things would stop working on one version of the compiler, then start working on the next.
It would also work on one person's machine but not on another.
So it became a problem to maintain.
As I have removed DOS support, I assume you are building with Makefile POSIX?
Actually the only issue with DOS lately is that its abstraction (which was based on POSIX) didn't have support for the latest directory changes, and this would explain why the "get first" and "get next" functions are not working correctly.
Now, why it is working on Win10? It makes no sense to me. If it is the same executable then it should work fine, unless the issue is with the operating system directory calls.

Cheers,
Marcelo.

guidol70, 2021-02-04:
Hi @MockbaTheBorg ,
I know the "DOS"-makefile-version is dead.
To compile for 32Bit Windows console I used the tdm-version of the Makefile
It also makes no sense to me that a RunCPM-binary does work (DIR-command) fine in Win10/WinXPSP3, but not via Extender in Win98SE/DOS

**But now something REAL strange :)**

I installed Visual Studio 2019 ( and YES it took around 6GB Diskspace :( ) and compiled RunCPM v4.8 with it.
While also getting a 32Bit RunCPM from Visual Studio I also had a chance to start it in Win98SE via KernelEX Kernel-Extensions,
**BUT** with this Visual Studio RunCPM v4.8 binary I **also do get this cut down** DIR-output :(
(and the binary is 16 times bigger than the version from tdm-gcc-32)

So it isnt the 16bit DJGPP and  not the compile via TDM-makefile because the problem is also with a Visual Studio compiled binary.

So no differnce in output between WinXPSP3/Win10 and Win98SE/DOS for a TDM-GCC or a Visual Studio binary.

It has - maybe - to be a output-handle problem from the DOS-Commandline. Even in GUI-Commandline or ins pure DOS commandline (original or 4DOS).

BTW: I couldnt start the Visual Studio 32bit binary via HX DOS Extender in DOS :(

Cheers,
Guido.

kklis, 2021-03-16:
DOS support was just an experiment. When I was porting an early version of RunCPM (AFAIR it was called CPMduino back then) to Posix (Linux, MacOSX) I added DOS routines simply because I could. DJGPP is largely GCC compatible and I did a lot of DOS programming back in the day when Turbo Assembler and Watcom ruled the Earth, so it was quite easy and fun. From a time perspective, I'm happy I didn't add an AmigaOS4 port, because I wanted to, and now it would just be another huge headache for Marcelo :) It was a good decision to phase out DOS support. It was just an experiment, and supporting it makes no sense any more. What's worse, it might require changes outside abstract.h and in consequence add unnecessary burden to the main code base. So don't bring it back. Just don't.