MockbaTheBorg/RunCPM #121: Issues on drive select

KamelBelgacem, 2021-02-16:
Hi @MockbaTheBorg,
I won't be bothering you with my thanks for the awesome revival you've done :)

After the commit of Feb 10 (RunCPM v5.0), I've discovered the following behaviour, using internal CCP (v1.7):

1. Can't switch drive when user area is not `0`: I have to go back to user area zero to switch each time
2. Randomly, when in drive `F5:` (for example), calling `TE XX.TXT` (for example) leads to errors then prompt becomes `A5:`! It seems that drive is switching to `A5:` to find `TE` then never goes back to F.. It's randomly happening.

Is issue coming from `cpm.h:L576` ?

I run this on an ESP32 DevKitC v2 (on a Minitel :))

MockbaTheBorg, 2021-02-16:
When I recently "fixed"  some things on the Internal CCP I was left with the impression that I might also had broken others.
You know the feeling when we leave the house but can't stop thinking we forgot something, then we keep checking wallet, car keys, etc.?

I have found that Internal CCP is actually deeply broken in terms of drive selection. I am wondering if there's a viable easy fix or if I should actually just write it from scratch (or drop it completely).

I was able to replicate these issues. I am now checking what would be the amount of effort to have them fixed.

Cheers,
Marcelo.

KamelBelgacem, 2021-02-16:
I've downloaded a version in January and it looks very stable (running on Raspbian 10), with Internal CCP v1.7. I didn't see any issue at all (RunCPM v4.8 I think). I was thinking to rollback the `cpm.h:L576` back but saw lots of other modifications on Feb 10, so wanted to check; may be a roll-back is just sufficient?

I didn't have to chance to find an operating CP/M 2.2 before RunCPM, but drive selection (globally) seems great, except when my Mix-C demos are in `C2` and compiler is in `C1`.. Path will check C2, A2, A0 (if I understood well), but never `C1` and calling `C2>c1:cc myfile.cc` doesn't work either. If this is standard CP/M behaviour then I think your iCCP is already great!

I let you work on it, I'll see on my side.

MockbaTheBorg, 2021-02-16:
I have just pushed a version with Internal CCP updated to v1.8.
I believe this one might have resolved the issues.
Please give it a try.

KamelBelgacem, 2021-02-16:
These are my current results with INTERNAL 1.8:

- Switching to different user areas and/drives without need to be on area zero: **OK**
- Calling external commands must not change current drive/area: **OK**
- Calling external commands from path different from file on parameters (`F2>B:TE C:HELLO.BAS`, etc.): **OK**

This one seems as stable as the Internal 1.6 (with "the feeling" it's better).

KamelBelgacem, 2021-02-16:
Found the following:

```
A0>stat
A: R/W, Space: 7464k

A0>b:
B0>c:
C0>a:
A0>stat
A: R/W, Space: 7464k
B: R/W, Space: 7464k
C: R/W, Space: 7464k

C0>
```

This forced the drive select to the latest visited one. But may be related to STAT.ASM..

MockbaTheBorg, 2021-02-16:
This seems like a different bug. It seems to be specific to Internal CCP. I will look at it.

MockbaTheBorg, 2021-02-17:
Internal CCP v1.9 seems to have fixed it, please check.

KamelBelgacem, 2021-02-17:
Drive select and STAT are **OK** for me.

MockbaTheBorg, 2021-02-17:
I have uploaded RunCPM v5.1 with CCP v2.0.
I did my best to test all I could see here, but please do a double check.
If working, please comment and close here.

drawkula, 2021-02-17:
One of my standard tests:
```
CP/M 2.2 Emulator v5.1 by Marcelo Dantas
      Built Feb 17 2021 - 20:11:36
-----------------------------------------
CCP : INTERNAL v2.0   CCP Address: 0xe400

RunCPM Version 5.1 (CP/M 2.2 60K)

A0>c:
C0>user 1
C1>dir *.sub
C: $$$      SUB  |  CR       SUB  |  MB6EL    SUB  |  MB6EL2   SUB
C: MBI      SUB  |  MYHI     SUB
C1>do mbi
SuperSUB V1.1

RunCPM Version 5.1 (CP/M 2.2 60K)

C1$; CREATE MBI.COM
C1$C MBI.C
HI-TECH C COMPILER (CP/M-80) V3.09
Copyright (C) 1984-87 HI-TECH SOFTWARE

RunCPM Version 5.1 (CP/M 2.2 60K)

C1$MBI
______________^^^^^^^^^^^^^]]]]]]]]]]]]]]]]]]]]]]]]]]]]]]]]]]]]^^^^^^^^^^^^^^^^
____________^^^^^^^^^]]]]]]]]]]]]]]]]]]]]]]]\\\\\\[[ZXWZZ[[\\\\\]]]]]^^^^^^^^^^
__________^^^^^^]]]]]]]]]]]]]]]]]]]]]]\\\\\\\\\\[[[ZYXWTJVYZ[[\\\\\\]]]]]]^^^^^
________^^^^^]]]]]]]]]]]]]]]]]]]]]\\\\\\\\\\[[[[ZZV@LO@@@BSSYZ[[[[\\\\\]]]]]]]^
_______^^^]]]]]]]]]]]]]]]]]]]]\\\\\\\\\[[ZZYYYYYYXWU@@@@@@FUWXYZZZZZZ[\\\]]]]]]
______^^]]]]]]]]]]]]]]]]]]\\\\\[[[[[[[ZZYWT@@@P@@@@@@@@@@@@@@@@PTUJRRHS[[\]]]]]
_____^]]]]]]]]]]]]]]]\\\[[[[[[[[[[ZZZZYUPRO@@@@@@@@@@@@@@@@@@@@@@@@@UXYZ[[\\]]]
____^]]]]]]\\\\\\\[[[ZZRVWXYXWIWXYYXXXU@E@@@@@@@@@@@@@@@@@@@@@@@@@@@@@KRZ[\\\]]
____]]\\\\\\\\[[[[[[ZYYXVN@@@@@@@@HRTT@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@SUZ[\\\]]
___]\\\\\\\[[ZZZZYXUTUTR@@@@@@@@@@@@@B@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@QXZ[[\\\\]
___XVURTT@OSPGOPMH@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@KTWYZZ[[\\\\]
___]\\\\\\\[[ZZZZYXUTUTR@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@RXZ[[\\\\]
____]]\\\\\\\\[[[[[[ZYYWVJ@@@@@@@@EQTT@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@QTZ[\\\]]
____^]]]]]]\\\\\\\[[[ZZQVWXXXVPWXYYXXXT@C@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@RZ[\\\]]
_____^]]]]]]]]]]]]]]]\\\[[[[[[[[[[ZZZZYSOQO@@@@@@@@@@@@@@@@@@@@@@@@MUXYZ[[\\]]]
______^^]]]]]]]]]]]]]]]]]]\\\\\[[[[[[[ZZYWT@@@P@@@@@@@@@@@@@@@@OTUIQQFV[[\]]]]]
_______^^^]]]]]]]]]]]]]]]]]]]]\\\\\\\\\[[ZYYYYYYXWWU@@@@@@CUWXYZZZZZZ[\\\]]]]]]
________^^^^^]]]]]]]]]]]]]]]]]]]]]\\\\\\\\\\[[[[ZZWALN@@@FSTYZ[[[[\\\\\]]]]]]]^
__________^^^^^^]]]]]]]]]]]]]]]]]]]]]]\\\\\\\\\[[[[ZYXWTNVYZ[[\\\\\\]]]]]]^^^^^
____________^^^^^^^^^]]]]]]]]]]]]]]]]]]]]]]]\\\\\\[[ZXWZZ[[\\\\\]]]]]]^^^^^^^^^
______________^^^^^^^^^^^^^]]]]]]]]]]]]]]]]]]]]]]]]]]]]]]]]]]]]^^^^^^^^^^^^^^^^

RunCPM Version 5.1 (CP/M 2.2 60K)

C1$;
C1>█
```
\o\   /o/   \o/
...and rebuilding SPL works again too...

Biiiiiig THANKS!

KamelBelgacem, 2021-02-17:
- STAT: **FIXED**
- DRIVE SELECT: **FIXED**

closed 2021-02-17 by KamelBelgacem