MockbaTheBorg/RunCPM #122: TYPE without parameters leads to random dumps

KamelBelgacem, 2021-02-16:
On RunCPM v5.0 using INTERNAL CCP v1.7 and v1.8, ESP32 board, calling TYPE without parameters leads to errors:

INTERNAL CCP 1.7 (**SD-Card was empty** except from drives and user areas):
```
A0>type
?   <--- this happens just after booting, when no 'user' command called previously
A0>user 3   <--- now type will always remember user area
A3>type
3?
A3>user 0
A0>type
0?
A0>
```

INTERNAL CCP 1.8: (after fixing drive select issue #121, with SD-Card full)
```
A0>TYPE
.           d($PRPR($PRK..     d($PRPR$(PRUMAKEFCB LIB d($  <-- lots of random memory dump
%I-feMLOAD    DOC XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
A0>user 9
A9> TYPE
!Rb£0  d"$PRPR"$PRVARUN   COM  <-- and again another type of memory blabla
A9>user 0
A0>type
.           d($PRPR($PRK..     d($PRPR$(PRUMAKEFCB LIB d($  <-- same dump from user 0
%I-feMLOAD    DOC XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
A0>b:
B0>type
 XXXXXXXXXXXXXxxx <--- same dump but with some files from local folder
```

For comparison, on RunCPM v4.8, INTERNAL CCP v1.6 running on Raspbian 10:
```
A0>type

A0>user 4
A4>type

A4>
```

I confirm that `TYPE` is dumping the content of the current directory tree with some extra.

Do you reproduce this?

MockbaTheBorg, 2021-02-17:
Yes ... I was able to reproduce it.
The single question mark (what happens after rebooting) is the correct behavior.
But after you type different commands, the buffer ends up dirty, and thus the error message from type gets messed.

I think I can fix it quickly.

MockbaTheBorg, 2021-02-17:
Internal CCP v1.9 fixed it, please check.

KamelBelgacem, 2021-02-17:
**Nothing changed** on my side.. Still the same results (tested with SDCard full of content).

I think there are additional broken stuff:
```
[RESET Board]
A0>o
O?
A0>ppp
PPP?

A0>4
4PP?
A0>you-going-nuts
YOU-GOING-NUTS?
A0>DIR
[dir results...]

A0>-me
-ME-GOING-NUTS?
```

KamelBelgacem, 2021-02-17:
- TYPE with no parameters must show "?": **NOT FIXED** (still a diarrhea of local catalog)
- Buffer of unknown commands clean (going-nuts thing): **FIXED**

drawkula, 2021-02-17:
I don't see that "(still a diarrhea of local catalog)" problem here (unchanged master branch):
```
A0>type

A0>█
```

KamelBelgacem, 2021-02-17:
If you don't reproduce it, I may be related to ESP32 and/or SDCard specific-case..
I'm going to give a try with a different CCP and see..
Anyone trying on a micro-controller?

KamelBelgacem, 2021-02-17:
I tried with DR and Z80 CCPs and the "diarrhea" happens again.
I close this and further investigation, may be related to ESP32 or my setup.

closed 2021-02-17 by KamelBelgacem