MockbaTheBorg/RunCPM #123: Internal CCP v1.9 vs SUBMIT

drawkula, 2021-02-17:
My typical HiTech-C workflow:
```
CP/M 2.2 Emulator v5.0 by Marcelo Dantas
      Built Feb 17 2021 - 04:55:05
-----------------------------------------
CCP : INTERNAL v1.9   CCP Address: 0xf400

A0>c:
C0>user 1
C1>type mbi.sub
; create mbi.com
c mbi.c
mbi
;

C1>do mbi
SuperSUB V1.1

C1$; CREATE MBI.COM
A1$C MBI.C
C?

A1$MBI
MBI?

A1$;
A1>█
```
While executing the `*.sub` lines, it falls back to drive A.

KamelBelgacem, 2021-02-17:
I'll try to reproduce this on ESP32 DevKitC v2 with SDCard, as well as all two issues "fixed" by Internal 1.9.

drawkula, 2021-02-17:
Rebuilding SPL via its `makespl.sub` file shows the same symptom:  Falling back to drive A while submit-ing.  So it's not a pure HiTech-C problem.
—▷ [SPL with its libs and demos (`MAKESPL.SUB`)](https://schorn.ch/cpm/zip/spl.zip)

An additional glitch:

On Debian...
```
(yeti@kumari:12)/home/wrk/runcpm/run$ ls -c H
2  1  0  F
```
Starting RunCPM in an own terminal...
```
CP/M 2.2 Emulator v5.0 by Marcelo Dantas
      Built Feb 17 2021 - 04:55:05
-----------------------------------------
CCP : INTERNAL v1.9   CCP Address: 0xf400

A0>user 3
A3>h:
H3>dir
No file
H3>█
```
Meanwhile looking at the Unix filesystem...
```
(yeti@kumari:12)/home/wrk/runcpm/run$ ls -c H
2  1  0  F
```
...no additional `user 3` subdir.

I noticed that when I tried to make a `test.sub` file in there but `te` was unable to save it.

KamelBelgacem, 2021-02-17:
I confirm it:

```
C1>TYPE HELLO.SUB
; CREATE HELLO.COM
CC HELLO.C
HELLO
;

C1>SUBMIT HELLO.SUB
C1$; CREATE HELLO.COM
A1$CC HELLO
CC?
A1$HELLO
HELLO?

A1$;
```

So yes, something is not going well for 1.9.
By the way, what I discovered in #122 (he-is-going-nuts etc.) cleans the buffer when calling SUBMIT:

```
C1>ppppp
PPPPP?
C1>X
XPPPP?
C1>SUBMIT HELLO
[as previous test, I see 'CC?', 'HELLO?' but not 'CCPPP?']
C1>X
X?
```

drawkula, 2021-02-17:
RESET!

My local clone might have been screwed.  I restarted with a fresh checkout of `master` and `te` could create the `test.sub` file in `H3:`.

Digging deeper...

KamelBelgacem, 2021-02-17:
Me too, an SDCard already broken (became empty under ESP32, all drives are accessible but nothing within. While PC/Mac never able to read/see the card).

I tried to remove the ";" sections of the .SUB file, but that made an infinite loop of `A1$CC HELLO.C` with blocks of strange characters. The first was `C1$CC HELLO.C` but did never produce a .COM, then lots of `A1$CC HELLO.C`.

Is it an issue within SUBMIT?

drawkula, 2021-02-17:
> Is it an issue within SUBMIT?

There was a time when it did not work.
There was a time when it did work.
...
`submit` has it's history in the closed issues.

drawkula, 2021-02-17:
```
CP/M 2.2 Emulator v5.0 by Marcelo Dantas
      Built Feb 17 2021 - 12:46:29
-----------------------------------------
CCP : CCP-CCPZ.64K   CCP Address: 0xf400
```
...has that `A:` submit problem too.
Adapting the title line...

MockbaTheBorg, 2021-02-17:
Ok ... the problem here is that the Internal CCP is (was always) intrinsically broken.
I don't want to get rid of it, because it actually brings a few advantages, but in order to make it independent of the RunCPM code I need to get rid of the cDrive and oDrive variables, which are kinda everywhere on RunCPM code.
This will require me to mess deeply with RunCPM, so brace yourselves, it is going to be a rough ride.
I will release a new RunCPM with the changes soon.

KamelBelgacem, 2021-02-17:
Wow! Why don't you tag stable releases so we can get them when needed?
I'm building a CP/M system so need something stable :)
I also like that Internal CCP because I intend to go berserk on it and add HTTP support (okay okay I know...)!
It also has "CLS" 😄

Hookay! I'll prepare my testing bench for it.
Good luck!

drawkula, 2021-02-17:
> It also has "CLS" smile

```
$ tput clear | od -c
0000000 033   [   H 033   [   J
0000006
```
A `cls.com` would be really easy if we can always assume a VT100 like terminal...
```
C1>type cls.c
#include "stdio.h"

main()
{
        write(1,"\033[H\033[J",6);
}
C1>rstat cls.*

 Recs  Bytes  Ext Acc
    1     1k    1 R/W C:CLS.C
   96    12k    1 R/W C:CLS.COM

C1>█
```
...but 12k?
```
C1>rstat clr.*

 Recs  Bytes  Ext Acc
    1     1k    1 R/W C:CLR.COM
    2     1k    1 R/W C:CLR.Z80

C1>type clr.z80
       ORG    0100H

       LD     DE,MSG
       LD     C,09H
       JP     0005H

MSG    DB     27,"[","H",27,"[","J"
       DB     "$"

C1>█
```
Better... ​o;-)

KamelBelgacem, 2021-02-17:
I thought about it.. but said that the `A0` will end full with all tools..

I tried it on the Minitel:

```
       ORG    100H
       LD     DE, MSG
       LD     C,9H
       JP     5H

MSG    DB     27,"[H",27,"[2J$"
```

It weights 16 bytes.
Thanks @drawkula 

BTW @MockbaTheBorg : I was happy that in RunCPM v5.0, I could use keyboard arrows (`\e[A`, etc.) in Multiplan & SuperCalc2 instead of CTRL+E/S/D/X, which wasn't the case on Run

MockbaTheBorg, 2021-02-17:
I have uploaded RunCPM v5.1 with CCP v2.0.
I did my best to test all I could see here, but please do a double check.
If working, please comment and close here.

MockbaTheBorg, 2021-02-17:
> I thought about it.. but said that the `A0` will end full with all tools..
> 
> I tried it on the Minitel:
> 
> ```
>        ORG    100H
>        LD     DE, MSG
>        LD     C,9H
>        JP     5H
> 
> MSG    DB     27,"[H",27,"[2J$"
> ```
> 
> It weights 16 bytes.
> Thanks @drawkula
> 
> BTW @MockbaTheBorg : I was happy that in RunCPM v5.0, I could use keyboard arrows (`\e[A`, etc.) in Multiplan & SuperCalc2 instead of CTRL+E/S/D/X, which wasn't the case on RunCPM 4.8. 'Hope you don't break that one 😆

I do not think there's a correlation between control keys working or not working between v4.8 and v5.0. I don't remember changing anything that would interfere on that.

MockbaTheBorg, 2021-02-17:
> Wow! Why don't you tag stable releases so we can get them when needed?
> I'm building a CP/M system so need something stable :)
> I also like that Internal CCP because I intend to go berserk on it and [...]

LOL ... there's no such thing as a "Stable RunCPM". It all depends on how I wake up that day ... 🤣

drawkula, 2021-02-17:
One of my standard tests:
```
CP/M 2.2 Emulator v5.1 by Marcelo Dantas
      Built Feb 17 2021 - 20:11:36
-----------------------------------------
CCP : INTERNAL v2.0   CCP Address: 0xe400

RunCPM Version 5.1 (CP/M 2.2 60K)

A0>c:
C0>user 1
C1>dir *.sub
C: $$$      SUB  |  CR       SUB  |  MB6EL    SUB  |  MB6EL2   SUB
C: MBI      SUB  |  MYHI     SUB
C1>do mbi
SuperSUB V1.1

RunCPM Version 5.1 (CP/M 2.2 60K)

C1$; CREATE MBI.COM
C1$C MBI.C
HI-TECH C COMPILER (CP/M-80) V3.09
Copyright (C) 1984-87 HI-TECH SOFTWARE

RunCPM Version 5.1 (CP/M 2.2 60K)

C1$MBI
______________^^^^^^^^^^^^^]]]]]]]]]]]]]]]]]]]]]]]]]]]]]]]]]]]]^^^^^^^^^^^^^^^^
____________^^^^^^^^^]]]]]]]]]]]]]]]]]]]]]]]\\\\\\[[ZXWZZ[[\\\\\]]]]]^^^^^^^^^^
__________^^^^^^]]]]]]]]]]]]]]]]]]]]]]\\\\\\\\\\[[[ZYXWTJVYZ[[\\\\\\]]]]]]^^^^^
________^^^^^]]]]]]]]]]]]]]]]]]]]]\\\\\\\\\\[[[[ZZV@LO@@@BSSYZ[[[[\\\\\]]]]]]]^
_______^^^]]]]]]]]]]]]]]]]]]]]\\\\\\\\\[[ZZYYYYYYXWU@@@@@@FUWXYZZZZZZ[\\\]]]]]]
______^^]]]]]]]]]]]]]]]]]]\\\\\[[[[[[[ZZYWT@@@P@@@@@@@@@@@@@@@@PTUJRRHS[[\]]]]]
_____^]]]]]]]]]]]]]]]\\\[[[[[[[[[[ZZZZYUPRO@@@@@@@@@@@@@@@@@@@@@@@@@UXYZ[[\\]]]
____^]]]]]]\\\\\\\[[[ZZRVWXYXWIWXYYXXXU@E@@@@@@@@@@@@@@@@@@@@@@@@@@@@@KRZ[\\\]]
____]]\\\\\\\\[[[[[[ZYYXVN@@@@@@@@HRTT@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@SUZ[\\\]]
___]\\\\\\\[[ZZZZYXUTUTR@@@@@@@@@@@@@B@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@QXZ[[\\\\]
___XVURTT@OSPGOPMH@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@KTWYZZ[[\\\\]
___]\\\\\\\[[ZZZZYXUTUTR@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@RXZ[[\\\\]
____]]\\\\\\\\[[[[[[ZYYWVJ@@@@@@@@EQTT@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@QTZ[\\\]]
____^]]]]]]\\\\\\\[[[ZZQVWXXXVPWXYYXXXT@C@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@RZ[\\\]]
_____^]]]]]]]]]]]]]]]\\\[[[[[[[[[[ZZZZYSOQO@@@@@@@@@@@@@@@@@@@@@@@@MUXYZ[[\\]]]
______^^]]]]]]]]]]]]]]]]]]\\\\\[[[[[[[ZZYWT@@@P@@@@@@@@@@@@@@@@OTUIQQFV[[\]]]]]
_______^^^]]]]]]]]]]]]]]]]]]]]\\\\\\\\\[[ZYYYYYYXWWU@@@@@@CUWXYZZZZZZ[\\\]]]]]]
________^^^^^]]]]]]]]]]]]]]]]]]]]]\\\\\\\\\\[[[[ZZWALN@@@FSTYZ[[[[\\\\\]]]]]]]^
__________^^^^^^]]]]]]]]]]]]]]]]]]]]]]\\\\\\\\\[[[[ZYXWTNVYZ[[\\\\\\]]]]]]^^^^^
____________^^^^^^^^^]]]]]]]]]]]]]]]]]]]]]]]\\\\\\[[ZXWZZ[[\\\\\]]]]]]^^^^^^^^^
______________^^^^^^^^^^^^^]]]]]]]]]]]]]]]]]]]]]]]]]]]]]]]]]]]]^^^^^^^^^^^^^^^^

RunCPM Version 5.1 (CP/M 2.2 60K)

C1$;
C1>█
```
\o\   /o/   \o/
...and rebuilding SPL works again too...

Biiiiiig THANKS!

closed 2021-02-17 by drawkula

MockbaTheBorg, 2021-02-19:
CCPZ actually has a SUBMIT issue.
It must use SUBMITD.COM instead of SUBMIT.COM (I have added a note to the
readme).

It can actually be easily patched. Just edit the CCPZ.60K (and/or 64K) file
with a Hex Editor, look for $$$ SUB and replace the 0x00 byte immediately
preceding it with 0x01.
This "fixes" CCPZ and makes SUBMIT work with it again.

There's some comment about this on the 1STREAD.ME file from disk A:

Cheers,
Marcelo

drawkula, 2021-02-19:
They already look patched in the repo?
```
(yeti@kumari:48)/home/wrk/runcpm/src/RunCPM/CCP$ head --bytes=144 CCP-CCPZ.60K | od -c
0000000 303 004 345 303  \0 345   P  \0                                
0000020                              \0           C   C   P   Z   -   V
0000040   4   .   1       o   f       1   1   /   2   7   /   8   1    
0000060      \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0
0000100  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0
0000120  \0  \0  \0  \0  \0  \0  \0  \0  \0  \b 344 027 344  \0  \0  \0
0000140  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0
0000160  \0  \0  \0  \0  \0  \0  \0   C   O   M 001   $   $   $        
0000200               S   U   B  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0
0000220
(yeti@kumari:48)/home/wrk/runcpm/src/RunCPM/CCP$ head --bytes=144 CCP-CCPZ.64K | od -c
0000000 303 004 365 303  \0 365   P  \0                                
0000020                              \0           C   C   P   Z   -   V
0000040   4   .   1       o   f       1   1   /   2   7   /   8   1    
0000060      \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0
0000100  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0
0000120  \0  \0  \0  \0  \0  \0  \0  \0  \0  \b 364 027 364  \0  \0  \0
0000140  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0
0000160  \0  \0  \0  \0  \0  \0  \0   C   O   M 001   $   $   $        
0000200               S   U   B  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0
0000220
```

MockbaTheBorg, 2021-02-19:
Sorry, my confusion, it is CCP_Z80 that needs the patch.
CCPZ should not need any patch.

So I am confused.

Is there an outstanding issue on CCPZ?

(edit) I have seen a forever-loop of submit happen once, and it seemed to
be related to the .SUB file having extra empty lines at the end. Can you
confirm?

drawkula, 2021-02-19:
> Is there an outstanding issue on CCPZ?

I don't remember current CCPZ problems, have switched to InternalCCP a while ago.

> I have seen a forever-loop of submit happen once, and it seemed to
be related to the .SUB file having extra empty lines at the end. Can you
confirm?

There was... https://github.com/MockbaTheBorg/RunCPM/issues/38#issuecomment-249745797

I don't remember what I had... that empty line error or the loop... too long ago... but I remember to have checked for not adding an empty line back then.