MockbaTheBorg/RunCPM #124: Nothing is working

erikbartmann, 2021-03-12:
Hello,

I started to upload the sketch on my Due, but many declarations are missing.

- SD
- LED
- LEDinv
- BOARD
- and so on...

Regards

Erik

MockbaTheBorg, 2021-03-12:
Hi Erik,

This is very strange.
Looks like some declaration is missing.
Please check the source code as, to use one or another hardware, a board
definition is needed in the main .ino file.
If you made the change and still the problems persisted, please let me know.

Thanks!
Marcelo

Also: consider joining our Discord channel, there we can have more "real
time" support and also easily share screenshots.

erikbartmann, 2021-03-12:
Hi Marcelo,

 

thanks for the quick answer!

 

I’m using the Arduino-Due, but the is no hardware-definition in the folder.

The following command rises an error.

 

#include "hardware/arduino/due.h"

 

Inside this folder, I can see only the subfolder avr.

 

 

 

Sorry for my stupid questions! I’m making a little mistake, but I don’t know what L

 

Regards

 

Erik

MockbaTheBorg, 2021-03-12:
Hi Erik,

There's no such thing as a stupid question. 🙂

I am talking about the RunCPM hardware folder. You need to download the entire project from GitHub, maybe you downloaded only the .ino file?
The whole folder containing the RunCPM.ino must be there, with the hardware subfolder, the .h file, everything.
If you are like me, who tries to skip the readme and go straight to building, there's no escape. For RunCPM there are a few details scattered on the readme file that are mandatory. Especially when you finally build and get to the point of building the CP/M folders, for example.
I am sure that it is some detail missing, we will figure it out. 

Cheers,
Marcelo

erikbartmann, 2021-03-12:
Hi Marcelo,

 

I think I got it – nearly J

 

I had to use the file 

 

due_sd_tf.h

 

instead of 

 

due.h

 

But the next issue ist he errormessage, that analogWrite was not declard in this scope.

Normally the Arduino.h file was missing. But it didn’t help.

 

Regards

 

Erik

erikbartmann, 2021-03-12:
Hi Marcelo,

I solved the analogWrite-Issue and ran into the next,

[image]

MockbaTheBorg, 2021-03-12:
This is one that's on the readme too.
As I said: in case of RunCPM, the GitHub readme is inescapable.
I recommend reading it fully, and only then start fighting the issues. Or you risk making some change to the code that will move farther from the solution, not closer. 

Cheers, 
Marcelo

erikbartmann, 2021-03-12:
You’re right! Thanks.

 

Erik

MockbaTheBorg, 2021-03-12:
Hi Erik,

Did it work at the end?

Thanks,
Marcelo.

erikbartmann, 2021-03-13:
Hi Marcelo,

 

yes it works and I feel bad about it because I asked you about such things that are in the description. 

RTFM is appropriate here. 

 

I just didn't look closely enough.

 

 

I apologize for that!

 

I have to thank you!

 

Erik

KamelBelgacem, 2021-03-15:
@erikbartmann In this case, are you okay to close the issue? :)

erikbartmann, 2021-03-15:
Thanks a lot!

Erik

closed 2021-03-15 by erikbartmann