MockbaTheBorg/RunCPM #125: with "internal" CCP UNARC.COM (v1.6) doesnt extract .ARK archives

guidol70, 2021-04-27:
there seem to be a problem with the "internal" CCP:
Some users and I did try to extract [CPMUG046.ARK](https://www.z80cpu.eu/mirrors/www.retroarchive.org/cpm/cdrom/SIMTEL/CPMUG/CPMUG046.ARK)

But when using the **"internal" CCP** (from RunCPM v5.2 or older - like v4.8) 
UNARC only does LIST the content of the .ARK
We are using the "normal" UNARC.COM v1.6 from your A.ZIP BootDrive.

On the other hand there seems also some little problem while using **CCP-DR** or **CCP-Z80** 
(**60k or 64K**) when extracting the archive.

With these CCPs the extraction of the archive does start, but after some files UNARC tells us in RunCPM that the destination-drive is full :(

As destination drive I used a empty (directory) named J/0 (J: User 0) and the archive is on K: (source).

Could you please be so kind and take a look, if you could find a problem/error?

[image]

MockbaTheBorg, 2021-04-27:
Hmmm. Strange.
I mean: I expect internal CCP to have bugs, but CCP-DR should have worked.
I will check it today.

guidol70, 2021-04-27:
CCP-DR worked (like CCP-Z80) that it started to extract, but I dont know why we shouldnt have enough space to extract a Usergroup-"Disk"-Archive.

I testes it on TTGO VGA32 , but also on the Windows Version  and there is also enough space on the harddrive.

Thanks for looking into it ;)

MockbaTheBorg, 2021-04-27:
We have two problems here. 

1: Internal CCP is broken, I do not know where yet, I will check.
2: CPMUG046.ARK contains filenames with '/', which is an invalid character for a FAT filesystem. Therefore it fails to create the file and UNARC interprets the error as "Directory Full".

Solution a:  HEX Edit CPMUG046.ARK, change the file names inside it and extract.
Solution b: Have RunCPM change the file names to something else than '/' when recording to disk.
I prefer a: as I don't think RunCPM should make changes to filenames. But I might (not a promise) create a flag so people can choose the behavior they want.

I will close this one for now as Internal CCP is well known to be broken anyways and I will be working on it as time allows.
I do suspect it is related to Internal CCP interpretation (storing) of the command line passed. Let's see.

closed 2021-04-27 by MockbaTheBorg

MockbaTheBorg, 2021-04-27:
I think I fixed both ... please test. RunCPM 5.3 and Internal CCP 2.1.

guidol70, 2021-04-27:
OK ;)
for now I do switch to the CCP-Z80 , because it does also show the User-Number on the prompt.

Secondly - for testing - I pacthed the .ARK-Filenames via a Hex-Editor and now the archive can be extracted competely :)
At the first sight I didnt see the "/"-problem, but now its very clear for me :)

The patched .ARK is included in the .ZIP

[UG046PAT.zip](https://github.com/MockbaTheBorg/RunCPM/files/6385150/UG046PAT.zip)

guidol70, 2021-04-27:
> I think I fixed both ... please test. RunCPM 5.3 and Internal CCP 2.1.

WOW - you were to fast. As you wrote the "internal" is broken I didnt expected the reapair so fast.

First I did try to add the line you changed to my files (because I have changes in ccp.h and globals.h, but didnt worked for me - so I copied the new version of the 2 files (disk.h hadnt any change and was copied 1:1) and added my changes (little bit for optic in gobals.h and my "Reboot" in ccp.h) and now it does work ;)

[image]