MockbaTheBorg/RunCPM #126: After "internal CCP" reset/reload MBASIC doenst load again

guidol70, 2021-04-30:
Maybe a issue after the last internal CCP correction? - because in RunCPM v5.2 it did work to load MBASIC again after leaving it with "system" as exit command.

With v5.4 from today I got this problem :( on Win and VGA32 (ESP32)
[image]
[image]

With v5.3 I got a debug-ouput at the end of the void setup of the .ino which is displayed in v5.3 (so the terminal did close?) but in v5.4 this isnt shown anymore :(

[image]

guidol70, 2021-04-30:
seems to be the CCP change from 3 days ago :(
fixed .ARK/UNARC, but I dont know wher could be the problem

MockbaTheBorg, 2021-04-30:
Please check again ... the latest v5.4 should have fixed it.

Waffle2, 2021-04-30:
It would appear that you have fixed it.  This was a problem for me as well, using m80/l80 and ddt.  The latest code indeed seems to solve it.  Thank you very much for the great environment, as well!

closed 2021-04-30 by MockbaTheBorg