MockbaTheBorg/RunCPM #127: How about RunMPM

snakebyte69, 2021-05-03:
An MP/M version will multiple serial ports would be awesome, something that could run 4 dumb terminals

MockbaTheBorg, 2021-05-03:
I thought about it long ago but I didn't see a use for it.
RunCPM is already "multi user" on Windows, Linux and Mac, as you can run multiple sessions sharing the same filesystem.
On small boards like Arduino and ESP the problem is lack of RAM to run banked systems, so it could become a problem.
To emulate MP/M a new hardware layer would need to be written, as the RunCPM "hardware" is not capable of some of the features MP/M requires. It would probably need to be re-done from the ground up.
So I decided that it was not worth spending time on it but instead focus on improving RunCPM itself, sorry.

closed 2021-05-03 by MockbaTheBorg

drawkula, 2021-05-03:
...and you easily can run MP/M on e.g. a Pi0, Pi1, ...

Look for the AltairZ80 emulator from the SIMH project, either directly from their [git repo](https://github.com/simh/simh) or from [P. Schorn](https://schorn.ch/altair.html), who also has [lots of ZIPs with OSes](https://schorn.ch/altair_4.php) and applications for it, including MP/M.