MockbaTheBorg/RunCPM #130: Source Files to Run CCPZ

gd-99, 2021-07-09:
The attached zip file contains the source codes to split up the CCP and BDOS to enable the classical CP/M CCP, BDOS and BIOS setup.

To load this version I modified main.asm to include:
`	        org	$4000
BDOS_BIN:	
		incbin	"bdos.bin"
BDOS_LENGTH	equ $ - BDOS_BIN`
Right after the cbios.bin bit around line 127

I then modified commands.asm to copy bdos.bin to the correct are using:
`	; copy BDOS to RAM
	ld	hl, BDOS_BIN
	ld	de,	$E806					;ORG set in BDOS.ASM - amend to suit
	ld	bc, BDOS_LENGTH
	ldir`
This is placed in cmd_cpm: function just under the cbios loading.
Here is a zip file containing CCPZ, BDOS and CBIOS:
[CCPZ_Sources.zip](https://github.com/MockbaTheBorg/RunCPM/files/6791534/CCPZ_Sources.zip)

MockbaTheBorg, 2021-07-10:
Hi gd-99,

Thanks for the submitted files.
I am not following the idea.
CCPZ is already supported on RunCPM, just need to select it as the CCP on the globals file.
So I am not sure how these would be used.
Can you elaborate more on the use case?

Thanks a lot,
Marcelo.

gd-99, 2021-07-10:
Hi Marcelo, I'm very sorry, I have only gone and posted to the wrong place Doh!

I will close - please feel free to delete - and post to the correct place.

Apologies again,

Glyn.

closed 2021-07-10 by gd-99

MockbaTheBorg, 2021-07-10:
No worries. 🙂