MockbaTheBorg/RunCPM #131: [Question/Help?] cant compile aymore :(

guidol70, 2021-09-11:
The last time I did compile successful at the 31.08.2021
and now I I get every time the same errors while compiling :(

I tested older RunCPM-versions (which did compile in th past), different sdfat (2.04.-2.1.0)
and ESP32 Arduino Support 1.0.6 and 2.0.0 and older and newer FabGL-version, but I do get the follwing error - do you got a hint? Seems to be on another end where it fails?
(or should I completly resinstall the Arduino-IDE / HW-Support and Librarys?)

```
Arduino: 1.8.15 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

In file included from Q:\1A_RunCPM\RunCPM_v5_5_master_TTGO_VGA32\RunCPM-master\RunCPM_VGA32_v5_5_Patch\RunCPM_VGA32_v5_5_Patch.ino:61:0:

Q:\1A_RunCPM\RunCPM_v5_5_master_TTGO_VGA32\RunCPM-master\RunCPM_VGA32_v5_5_Patch\abstraction_arduino.h: In function 'bool _RamLoad(char*, uint16)':

abstraction_arduino.h:24:2: error: 'File' was not declared in this scope

  File f;

  ^

abstraction_arduino.h:27:6: error: 'f' was not declared in this scope

  if (f = SD.open(filename, FILE_READ)) {

      ^

Q:\1A_RunCPM\RunCPM_v5_5_master_TTGO_VGA32\RunCPM-master\RunCPM_VGA32_v5_5_Patch\abstraction_arduino.h: At global scope:

abstraction_arduino.h:38:1: error: 'File' does not name a type

 File rootdir, userdir;

 ^

abstraction_arduino.h:60:1: error: 'File' does not name a type

 File _sys_fopen_w(uint8* filename) {

 ^

abstraction_arduino.h:64:26: error: 'File' has not been declared

 int _sys_fputc(uint8 ch, File& f) {

                          ^

Q:\1A_RunCPM\RunCPM_v5_5_master_TTGO_VGA32\RunCPM-master\RunCPM_VGA32_v5_5_Patch\abstraction_arduino.h: In function 'int _sys_fputc(uint8, int&)':

abstraction_arduino.h:65:11: error: request for member 'write' in 'f', which is of non-class type 'int'

  return(f.write(ch));

           ^

Q:\1A_RunCPM\RunCPM_v5_5_master_TTGO_VGA32\RunCPM-master\RunCPM_VGA32_v5_5_Patch\abstraction_arduino.h: At global scope:

abstraction_arduino.h:68:18: error: variable or field '_sys_fflush' declared void

 void _sys_fflush(File& f) {

                  ^

abstraction_arduino.h:68:18: error: 'File' was not declared in this scope

abstraction_arduino.h:68:24: error: 'f' was not declared in this scope

 void _sys_fflush(File& f) {

                        ^

[... 346 lines cut ...]
           _sys_fflush(pun_dev);

                              ^

RunCPM_VGA32_v5_5_Patch:439:13: error: 'lst_dev' was not declared in this scope

         if (lst_dev)

             ^

RunCPM_VGA32_v5_5_Patch:440:30: error: '_sys_fflush' was not declared in this scope

           _sys_fflush(lst_dev);

                              ^

exit status 1

'File' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

guidol70, 2021-09-11:
Now Iam at Arduino-IDE v1.8.6, SDFat v2.1.0, ESP HW Support v1.0.6
and
FabGL-github-master from the 02.08.2021

And now RunCPM v5.6 does compile - but a newer FabGL-master does give me errors... 
I will ask there :)

MockbaTheBorg, 2021-09-17:
Hi @guidol70 I see you are able to build correctly now?
Should we close this one?

guidol70, 2021-09-17:
> Hi @guidol70 I see you are able to build correctly now?
> Should we close this one?

We could close this ;)
Iam using the "old" /src/fabutils.cpp from FabGL 1.0.4 and the "old" ESP32-Core v1.0.6

closed 2021-09-19 by MockbaTheBorg