MockbaTheBorg/RunCPM #132: TXRX question

B1tbang3r, 2021-09-16:
Hello ive build runcpm and i verified that its working through the serial monitor in the arduino ide

i hooked up a serial terminal board (esp32 with fabgl) and tried rxtx pin 0 and 1 on the arduino due but nothing

ive set the correct baud settings i think 9600 8N1 

is there another pinout i should follow or is this  txrx not implememted and only accesible through the usb ports ?

guidol70, 2021-09-16:
@B1tbang3r 
Pin0 (RX0) ans Pin1 (TX0) is normally working on the DUE, but can interference when the USB-Port not only delivers power - but has a computer connected who do try to use the USB-serial port.
I had this problem in the past - see the following issue:
https://github.com/MockbaTheBorg/RunCPM/issues/117

So as you can read there is better to replace all "Serial." to "Serial1." in all tabs of the .ino and use Pin19 (RX1) and Pin18 (TX1)

The baudrate is OK (when not changed in the .ino) with 9600 Baud
```
// Serial port speed
#define SERIALSPD 9600
```

If you did receive anything is allways a good test to swap the RX/TX Lines....
Sometimes (depending on Hardware) you have to connect RX/TX and TX/RX or RX/RX and TX/TX ;)

MockbaTheBorg, 2021-09-17:
Hi @B1tbang3r . How did this go?
Were you able to achieve a solution?

B1tbang3r, 2021-09-17:
Hello , 

yes i was , turns out the arduino needed a good reset after power up :) 
really enjoying Runcpm just starting to get to know its quirks

closed 2021-09-19 by MockbaTheBorg