MockbaTheBorg/RunCPM #134: [Info] compile Problem in RunCPM for VGA32 between SDFat.h and FS.h

guidol70, 2021-09-21:
In the [Issue 131](https://github.com/MockbaTheBorg/RunCPM/issues/131) I had problem compiling RunCPM for VGA32 (ESP32) with the error-code
**'File' was not declared in this scope**

This did occur because FabGL is getting ready for ESP32 Core v2.0.0 and now
does include in **fabutils.cpp** the **vfs_api.h** which does include the **FS.h** (from the ESP32 Core)

Now here I do get a problem, because **SDFat.h** and **FS.h** do have the Keyword **File** :(

To get rid of the problem I searched through the GitHub-Issue of the SDFat-project of Greiman and did found the following issue:
[SdFat 2-beta to SdFat 2.0 release #302](https://github.com/greiman/SdFat/issues/302)

The solution is to use an "**alias**" for **File & the SDFat-Library**.
In **SDFat.h** is an definition that you could use **File32** as alias for **File**

```
The problem is File is defined by ESP32 in some cases.
File is just a typedef for File32.
The only safe thing to do on ESP32 is to use File32
For Fat16/FAT32 only use SdFat32 and File32.

```
So I replaced all "**File f;**" with "**File32 f;**" and also some/all **"File"-function-calls** on other places in the .ino and his tabs (where I did think its OK) :)

After that I could compile RunCPM for VGA32 with ESP32 Core v1.0.6 and the latest fabutils.cpp (using the vfs_api.h)
```
ESP32 Core v1.0.6:
==================
Using library SPI at version 1.0 in folder: C:\Users\guido\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\SPI 
Using library SdFat at version 2.1.0 in folder: C:\Users\guido\Documents\Arduino\libraries\SdFat 
Using library FabGL at version 1.0.4 in folder: C:\Users\guido\Documents\Arduino\libraries\FabGL 
Using library Preferences at version 1.0 in folder: C:\Users\guido\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\Preferences 
Using library FS at version 1.0 in folder: C:\Users\guido\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\FS 
Using library WiFi at version 1.0 in folder: C:\Users\guido\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi 
"C:\\Users\\guido\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-size" -A "C:\\Users\\guido\\AppData\\Local\\Temp\\arduino_build_364655/RunCPM_VGA32_v5_6_File32_Test.ino.elf"

Sketch uses 568766 bytes (43%) of program storage space. Maximum is 1310720 bytes.
Global variables use 84784 bytes (25%) of dynamic memory, leaving 242896 bytes for local variables. Maximum is 327680 bytes.
```
and it **worked like beforee with v1.0.6 :)**

Then I did compile with ESP32 Core v2.0.0:
```
ESP32 Core v2.0.0:
==================
Using library SPI at version 2.0.0 in folder: C:\Users\guido\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0\libraries\SPI 
Using library SdFat at version 2.1.0 in folder: C:\Users\guido\Documents\Arduino\libraries\SdFat 
Using library FabGL at version 1.0.4 in folder: C:\Users\guido\Documents\Arduino\libraries\FabGL 
Using library Preferences at version 2.0.0 in folder: C:\Users\guido\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0\libraries\Preferences 
Using library FS at version 2.0.0 in folder: C:\Users\guido\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0\libraries\FS 
Using library WiFi at version 2.0.0 in folder: C:\Users\guido\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0\libraries\WiFi 
"C:\\Users\\guido\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\gcc8_4_0-esp-2021r1/bin/xtensa-esp32-elf-size" -A "C:\\Users\\guido\\AppData\\Local\\Temp\\arduino_build_252968/RunCPM_VGA32_v5_6_File32_Test.ino.elf"
Sketch uses 606625 bytes (46%) of program storage space. Maximum is 1310720 bytes.
Global variables use 83756 bytes (25%) of dynamic memory, leaving 243924 bytes for local variables. Maximum is 327680 bytes.
```
but RunCPM for VGA32 **didnt start :(** its hanging in a **Reboot-Loop** with the following **message**:
```
Rebooting...
assertion "xTaskToNotify" failed: file "IDF/components/freertos/tasks.c", line 5308, function: xTaskGenericNotify
abort() was called at PC 0x4010e18b on core 0
Backtrace:0x40104ac1:0x3ffb26600x4008aa3d:0x3ffb2680 0x4009063d:0x3ffb26a0 0x4010e18b:0x3ffb2720 0x4008d8c1:0x3ffb2750 0x400f31b9:0x3ffb2770 0x400f32af:0x3ffb27a0 0x400e05c1:0x3ffb27c0 0x400f92e3:0x3ffb2820 
ELF file SHA256: 0000000000000000

```
It would be nice if anybody got an idea why this freertos-problem does occour ;)

MockbaTheBorg, 2021-09-22:
I see there's a lot happening on the VGA32 thread about this. Looking forward to see you succeed.

Cheers,
Marcelo.

guidol70, 2021-09-25:
> I see there's a lot happening on the VGA32 thread about this. Looking forward to see you succeed.

I do my very best, but I cant find the difference (in base functions) between the FabGL ANSITerminal and "my" RunCPM - with ESP32 Core v1.0.6 its running and with ESP32 Core v2.0.0 I didnt get input from the keyboard (but RunCPM is running when controlled from USB-serial - so Terminal, SD, USB is running. Only the PS2Controller doenst like me ;) )
I did try MANY different way of source and compiling - now I will take a break. 
Maybe waiting for ESP32 Core v2.0.1? :)

MockbaTheBorg, 2021-09-26:
Maybe the esp core folks will give something.
Looks like they are going to try your code, no?

guidol70, 2021-09-26:
> Maybe the esp core folks will give something. Looks like they are going to try your code, no?

I  dont think so ;)
But the only thing that changed is the ESP32 Core - and for v1.x they produced also many versions (up to 1.0.6)

MockbaTheBorg, 2021-10-03:
I will close this issue then, as we have determined the issue belongs with VGA32.
I hope they resolve it.

closed 2021-10-03 by MockbaTheBorg