MockbaTheBorg/RunCPM #140: VT340 feature request

snakebyte69, 2021-12-04:
How about expanding the terminal capability from VT-100 to VT-340 for more graphic capabilities?

drawkula, 2021-12-04:
...like this?

[image]

[MB6EL.ZIP](https://github.com/MockbaTheBorg/RunCPM/files/7654885/MB6EL.ZIP)

snakebyte69, 2021-12-04:
How are you running that?

drawkula, 2021-12-05:
I'm using an XTerm with enabled VT340 features.  My `~/.Xresources` contains:
```
!! for SIXELS
!!
XTerm*decTerminalID: vt340
XTerm*numColorRegisters: 256
```

snakebyte69, 2021-12-07:
Have you tried it on a Mac?

drawkula, 2021-12-07:
No Apple stuff here, except the ones that really grow on trees.

drawkula, 2021-12-07:
Related:
—▷ https://github.com/MockbaTheBorg/RunCPM/issues/141

MockbaTheBorg, 2021-12-07:
Yes ... this is a terminal capability "issue", not RunCPM capability. There's nothing for RunCPM to do here. I will close as it is (technically) already implemented.

closed 2021-12-07 by MockbaTheBorg