MockbaTheBorg/RunCPM #141: Programmable Character Graphics

snakebyte69, 2021-12-07:
How about user specified fonts allowing for programmable characters so PETSCII graphics and systems like the Microbee could be emulated. A font for Mac/Windows could be created with PCG characters that match what is running on the target machine. CP/M development could be done running under RunCPM modified for the target machine when finished.

drawkula, 2021-12-07:
Why?

CP/M systems are not bound to one terminal type.  Serious CP/M software ™should be™ configurable for a variety of terminal types.

RunCPM does not include the terminal emulation.  It just communicates to an external terminal.

Unicode?

That'd be sloooooow.  How would you fit the bitmaps of tens of thousands of characters into a 64kByte system?  The character definitions would need to be on the mass storage instead.

Emulations for systems with only one terminal type with graphics (like Sinclair Spectrum) may fit these ideas (except Unicode) better.

guidol70, 2021-12-07:
> Why?
> 
> CP/M systems are not bound to one terminal type. Serious CP/M software ™should be™ configurable for a variety of terminal types.
> 
> RunCPM does not include the terminal emulation. It just talks to an external terminal.
> 

but you can configure puTTY to use a C64 PETSCII-TrueType-Font ;)

[image]

drawkula, 2021-12-07:
That does not contradict what I wrote.
It even underlines it!
You changed the font the terminal(-emulator-program) uses, not something in RunCPM.

MockbaTheBorg, 2021-12-07:
But what I understood from @snakebyte69 is that he is thinking about a custom made TTF font to use on the terminal emulator (Windows, Linux or Mac) to emulate those systems. Not adding a font support to RunCPM, am I correct?

snakebyte69, 2021-12-08:
> But what I understood from @snakebyte69 is that he is thinking about a custom made TTF font to use on the terminal emulator (Windows, Linux or Mac) to emulate those systems. Not adding a font support to RunCPM, am I correct?

Yes. Commodore PET is PCG only, Microbee has PCG capability. It's not about speed it's about developing for old 8 bit systems using a Mac Laptop (in my case) and tools. Sublime Text, Cross Compilers, RunCPM, fonts as a substitute for PCG characters.

MockbaTheBorg, 2021-12-17:
I think this one is settled as we have established that the font support should be a functionality of the terminal server (or terminal emulator) being used.

closed 2021-12-17 by MockbaTheBorg