MockbaTheBorg/RunCPM #143: [Info] new error-message because of SDFatConfig.h

guidol70, 2022-04-01:
since the last ESP32-Core (v2.0.2) I had a new error message about the missing change for the SDFatConfig.h (because of a SDFat-Library Update to v2.1.2)
from #define SDFAT_FILE_TYPE 3
to #define SDFAT_FILE_TYPE 1

In the past we did get the message
'File' has no member named 'dirEntry'

But now I had the following error/messages:
- error: no match for 'operator=' (operand types are 'File32' and 'FsFile')
- note: candidate: 'File32& File32::operator=(const File32&)'
- class File32 : public StreamFile<FatFile, uint32_t

when compiling RunCPM for ESP32/VGA32

But with the normal change the compile does work again fine :)

MockbaTheBorg, 2022-04-01:
Interesting ... so what is the solution exactly? Not to make the change?

guidol70, 2022-04-02:
> Interesting ... so what is the solution exactly? Not to make the change?
No :) You have to make the change, but it was a new error message and I had to compare my complete Arduino IDE installation - only to find out that it was the missing change after a SDFat-Library update.

So when anyone is getting this "new" error he should think about the "old" change ;)

MockbaTheBorg, 2022-04-05:
So the solution is to revert? Back to original pre-change .h file?

MockbaTheBorg, 2022-04-05:
Strange that I upgraded my SdFat version to 2.1.2 (from 2.1.0) and I do not have any error?
Maybe this is happening to specific boards?  
I want to establish what would be the milestone to close this issue.

guidol70, 2022-04-06:
> So the solution is to revert? Back to original pre-change .h file?

No - the change is the same as on your READ.ME at github, but I did get the other error because of the missing change.
All was fine after applying the change like before when you got a new fresh SDFat-Library.

I dont know from where I did come, because in my portable Installation of the Arduino IDE I didnt get this error-message.

guidol70, 2022-04-06:
> Strange that I upgraded my SdFat version to 2.1.2 (from 2.1.0) and I do not have any error? Maybe this is happening to specific boards? I want to establish what would be the milestone to close this issue.

with applied change I didnt got also any problems with SDFat 2.1.2

MockbaTheBorg, 2022-04-09:
I think v5.7 already got the change. As I was able to upgrade normally from 2.1.1.
I will close for now and then we open one issue in case we find build problems on some platform.

closed 2022-04-09 by MockbaTheBorg