MockbaTheBorg/RunCPM #144: Redirect console input and output from/to files

pzembrod, 2022-05-05:
Hi MockbaTheBorg,

I started using RunCPM to automate builds and tests (via gnu-make) of [VolksForth](https://github.com/pzembrod/VolksForth/tree/cpm-tests/8080/CPM) and found that just redirecting stdin and stdout of RunCPM didn't play well with VolksForth, or vice versa. If my analysis is correct, then the key problem was around input keys became available slower through the Linux pipe than the guest CP/M and program consumed them. LF vs CR was a secondary problem, but that could have been solved.

In any case, I started experimenting with command line flags to read console input from and to log console output to a file, which ended up working nicely, with relatively little effort. You can take a look at my [changed code here](https://github.com/pzembrod/RunCPM), and also [how it's used](https://github.com/pzembrod/VolksForth/blob/cpm-tests/8080/CPM/emulator/run-in-runcpm.sh#L29).  I'm happy to send you a pull request, but I thought I'd open this issue first to discuss.

Cheers & thanks for RunCPM - I find it very handy
/Philip

MockbaTheBorg, 2022-05-07:
This is an interesting mod. 
I would gladly merge it onto the main code if we can get it to work seamlessly on the desktop platforms (Windows, Mac, Linux).
Have you joined the discord channel?
Maybe we can start discussing there and have the team test it on other platforms, then we come up with a common code.

Thanks,
Marcelo

MockbaTheBorg, 2022-05-12:
I will close this one as it is not a real issue. We will be discussing it on Discord.

closed 2022-05-12 by MockbaTheBorg