MockbaTheBorg/RunCPM #145: Two problems on macosx

cdhdds, 2022-05-08:
When trying to use any other CCP it does not load and aborts.  Have tried all by changing in globals.h.  Can tell it is supposed to load one by response:

CCP : CCP-Z80.60K   CCP Address: 0xe400

RunCPM Version 5.7 (CP/M 2.2 60K)
Unable to load CP/M CCP.
CPU halted.

*********************************************

When using the internal CCP which is all I have been able to use, SAVE does not work on macosx.  The same happens on Windows with internal CCP but with CCP-DR works fine but as above cannot load CCP-DR on macosx.  Output:

C0>SAVE 5 TEST

C0>dir
C: 5

C0>TYPE 5

C0>

cdhdds, 2022-05-13:
> Thanks for catching this.
> Should be resolved now. Please download the latest and test.
>
> Marcelo.

Marcelo,

Fix works fine.  Thanks.  Would it be possible for someone to give me the address of the CLS function in the internal CCP?  Would be helpful in coding in 8080 assembler.

MockbaTheBorg, 2022-05-13:
The internal CCP doesn't run as native Z80 code, therefore is not addressable from the emulator perspective (from Z80 code space).
To clear the screen, the best options is to set up a clear-screen string and print it out using a regular CP/M BDOS ball.
Either that or switch to a native CCP.