MockbaTheBorg/RunCPM #147: [Question] Different Upper-/Lowercase Filename handling in CCPs?

guidol70, 2022-05-27:
I would like to know if there is another handling in converting Upper-/Lowercase Filenames in the internal CCP (also v2.4) against the CCP-ZCP3?

I know normally, I should have all filenames in Uppercase (also do you MultiCommander for Batch rename under Windows - which doesnt work recursively) but also when they are all Uppercase sometime its more easy to type the filename in lowercase.

With the Internal-CCP the follwing **does work**:
mbasic fractal.bas
but inside MBASIC the following **doenst work**:
load "fractal.bas"
but it works with
load "FRACTAL.BAS"

Today I did try the ZCP3-CCP and was wondering that I did get FRACTAL.BAS loaded inside
MBASIC with:
load "fractal.bas" (and also load "FRACTAL.BAS" or "FRACTAL.bas" **worked**)
So complete working conversion to Uppercase here.

Incidentally I also got a mixed up Filename like FRACTAL**.pas (lowercase-extension)** and PIP couldnt find it when I did a PIP B:FRACTAL.PAS=A:FRACTAL.PAS
But this also happens with the ZCP3-CCP and PIP. **I think thats Ok**, because PIP is self-handling the filenames - so it does a uppercase on all chars and couldnt find the FRACTAL.pas

It also cant find it, when named "right":
```
RunCPM [v5.7]  CCP:[CCP-ZCP3.64K]  TPA:[64K]
B1>a0:PIP LOW.LOW=FRACTAL.pas
NO FILE: =FRACTAL.PAS

```

ratboy666, 2022-05-27:
With CP/M 2.2 CCP, the CCP uppercases command lines. Not just filenames
-- there really is no way
to enter lowercase into the command line.

MBASIC, does NOT convert strings to uppercase. The BDOS does not
convert filenames in FCBs at
all.

So
mbbasic
load "fractal.bas"

is different from
load "FRACTAL.BAS"

As I only used CP/M 2.2 "in the day", this behaviour is expected.

If you ended up with a filename with lower-case characters, they ended
up "difficult" to handle.
PIP, for example, converts commands to uppercase, so cannot be used.
REN and ERA cannot be used.
MBASIC to the rescue. Since MBASIC does NOT change its string arugment,
it can be used to delete

kill "lower.fil"

Indeed, any characters can be used:

kill "'.bas"

to remove '.bas

Still, very difficult to deal with files with : or . in the name! (that
parsing IS done by mbasic).

MockbaTheBorg, 2022-05-28:
Initially, when I started writing RunCPM I thought of messing with Upper/Lowercase myself, but then I decided to not do it and interfere minimally (if at all) with the original software.
So what you see happening now is what I remember seeing back in the days when I used my AppleII.
In my opinion this adds to the "historically correct experience". :)

closed 2022-05-28 by MockbaTheBorg