MockbaTheBorg/RunCPM #149: missing functions for CCP v2.5 in abstraction_arduino.h?

guidol70, 2022-07-23:
I did try to compile the internal CCP v2.5 for "Arduino" here VGA32, but the functions used in the new internal CCP v2.5 are (from my sight) are only in **abstraction_posix.h** - like 

- _sys_exists
- _sys_fopen_r
- _sys_fopen_w
- _sys_fread
- _sys_readseq

and not in **abtractions_arduino.h**

Error-Message from my compile:

```
In file included from Q:\1A_RunCPM\RunCPM_v5_7_master_TTGO_VGA32\RunCPM-master\RunCPM_VGA32_v5_7_CCP25_23072022\RunCPM_VGA32_v5_7_CCP25_23072022.ino:138:
Q:\1A_RunCPM\RunCPM_v5_7_master_TTGO_VGA32\RunCPM-master\RunCPM_VGA32_v5_7_CCP25_23072022\ccp.h: In function 'uint8 _ccp_vol()':
ccp.h:388:7: error: '_sys_exists' was not declared in this scope
   if (_sys_exists(folder)) {
       ^~~~~~~~~~~
Q:\1A_RunCPM\RunCPM_v5_7_master_TTGO_VGA32\RunCPM-master\RunCPM_VGA32_v5_7_CCP25_23072022\ccp.h:388:7: note: suggested alternative: '_sys_errlist'
   if (_sys_exists(folder)) {
       ^~~~~~~~~~~
       _sys_errlist
ccp.h:394:18: error: '_sys_fopen_r' was not declared in this scope
     FILE* file = _sys_fopen_r(filename);
                  ^~~~~~~~~~~~
Q:\1A_RunCPM\RunCPM_v5_7_master_TTGO_VGA32\RunCPM-master\RunCPM_VGA32_v5_7_CCP25_23072022\ccp.h:394:18: note: suggested alternative: '_sys_fopen_w'
     FILE* file = _sys_fopen_r(filename);
                  ^~~~~~~~~~~~
                  _sys_fopen_w
ccp.h:395:24: error: '_sys_fread' was not declared in this scope
     bytesread = (uint8)_sys_fread(&dmabuf[0], 1, 128, file);
                        ^~~~~~~~~~
Q:\1A_RunCPM\RunCPM_v5_7_master_TTGO_VGA32\RunCPM-master\RunCPM_VGA32_v5_7_CCP25_23072022\ccp.h:395:24: note: suggested alternative: '_sys_readseq'
     bytesread = (uint8)_sys_fread(&dmabuf[0], 1, 128, file);
                        ^~~~~~~~~~
                        _sys_readseq

```
Or missed I something else?

MockbaTheBorg, 2022-07-24:
I think it is resolved, please check it out.
So far I tested on the Teensy, but should work the same on others.

guidol70, 2022-07-25:
> I think it is resolved, please check it out. 

As you did wrote on Discord - compile does now work for Arduino/VGA32/RPi_Pico but without a working VOL command ;)

MockbaTheBorg, 2022-07-25:
I have pushed more changes and got it working here for me on the Teensy at least.
Can you check?

guidol70, 2022-07-25:
> I have pushed more changes and got it working here for me on the Teensy at least. Can you check?

Did check - does compile, but without showing the User-Area-Number like on you first Windows-Screenshot.
Picture could be seen at Discord-server

MockbaTheBorg, 2022-07-25:
Resolved on https://github.com/MockbaTheBorg/RunCPM/commit/711f06c56f513925ff3b6fec9e6dc0ba19a60ed7

closed 2022-07-25 by MockbaTheBorg