MockbaTheBorg/RunCPM #151: cpu.h "breaks" C++17 Standard with "register"-commands

guidol70, 2022-09-22:
while compiling for the Raspberry Pico I normally used the option **-std=gnu11 and -std=gnu++11**
because with the newer **-std=gnu17 and -std=gnu++17**
I do get the warnings as in the attached file

How "hard" does the compiler need these register lines, because as info from
https://stackoverflow.com/questions/3207018/register-keyword-in-c
```
register is a hint to the compiler, advising it to store that variable in a processor register instead of memory (for example, instead of the stack).

The compiler may or may not follow that hint.
```

Is there a way you know to get around these warnings with  **-std=gnu17 and -std=gnu++17**?

[C_plus_plus_17_warnings.txt](https://github.com/MockbaTheBorg/RunCPM/files/9625555/C_plus_plus_17_warnings.txt)

guidol70, 2022-09-22:
using **-Wno-register**
in the compile options with -std=gnu++17
does suppress the register-warnings and gnu++17 does "accept" the register-commands ;)

MockbaTheBorg, 2022-09-29:
If we rebuild for all the GNU platforms with the new options, without any side effects, I will gladly accept a pull-request on that.

MockbaTheBorg, 2022-10-03:
Thinking a little bit about this, the "register" could make a counter be on a CPU register ad thus speed up the execution. How great would be the gain? No idea, we would have to measure.
I am not sure if the compilers nowadays, with all the smart optimizations, woudn't be already taking advantage of this regardless.
I suggest a performance test on big and small processors, with and without "register" to see if it makes a difference.

MockbaTheBorg, 2022-10-03:
Tested with Visual Studio and (as expected) didn't notice any performance issues. "register" removed.

closed 2022-10-03 by MockbaTheBorg