MockbaTheBorg/RunCPM #154: [PR #152] xxx.sub == xxx.com now

drawkula, 2022-10-01:
```
C1>dir myhi*.*
C: MYHI     C    |  MYHI     COM  |  MYHI     SUB
C1>type myhi.c
#include "stdio.h"
main()
{
        puts("No meltdown, no specte.");
        puts("No rowhammer, no brainslammer!");
        puts("\nCP/M!");
        puts(";-)");
}

C1>myhi.sub
No meltdown, no specte.
No rowhammer, no brainslammer!

CP/M!
;-)

C1>type myhi.sub
; create myhi.com
c myhi.c
myhi
;

C1>█
```
`C1>myhi.sub` seems to have executed the `.COM` file.
Is that intended and I just don't understand something?

drawkula, 2022-10-01:
It should have compiled the source by `c myhi.c` which would not have been silent, before starting the resulting COM file if it would have called the subfile correctly.
That would look like:
```
C1>do myhi
SuperSUB V1.1

C1$; CREATE MYHI.COM
C1$C MYHI.C
Hi-Tech Z80 C Compiler (CP/M-80) V3.09-16
Copyright (C) 1984-87 HI-TECH SOFTWARE
Updated from https://github.com/agn453/HI-TECH-Z80-C

C1$MYHI
No meltdown, no specte.
No rowhammer, no brainslammer!

CP/M!
;-)

C1$;
C1>▙
```
After removing `myhi.com` the extension-less command is interpreted as subfile:
```
C1>era myhi.com
C1>myhi

C1$; CREATE MYHI.COM
C1$C MYHI.C
Hi-Tech Z80 C Compiler (CP/M-80) V3.09-16
Copyright (C) 1984-87 HI-TECH SOFTWARE
Updated from https://github.com/agn453/HI-TECH-Z80-C

C1$MYHI
No meltdown, no specte.
No rowhammer, no brainslammer!

CP/M!
;-)

C1$;
C1>█
```
Same with `.SUB` extension if the COM file is not there:
```
C1>era myhi.com
C1>myhi.sub

C1$; CREATE MYHI.COM
C1$C MYHI.C
Hi-Tech Z80 C Compiler (CP/M-80) V3.09-16
Copyright (C) 1984-87 HI-TECH SOFTWARE
Updated from https://github.com/agn453/HI-TECH-Z80-C

C1$MYHI
No meltdown, no specte.
No rowhammer, no brainslammer!

CP/M!
;-)

C1$;
C1>█
```
Now `myhi.com` and `myhi.sub` exist:
```
C1>myhi.sub
No meltdown, no specte.
No rowhammer, no brainslammer!

CP/M!
;-)

C1>█
```
No sign of compiler output.

geowar1, 2022-10-02:
What’s the behavior using SUBMIT command (with & without my PR)?

MockbaTheBorg, 2022-10-02:
I see Drawkula's point. He explicitly called MyHi.SUB, but this executes the .COM. We probably shouldn't be overriding the file extension without checking first.

geowar1, 2022-10-02:
Ah! I think I got it… Yeah, I’ll check the extension first. Good catch. ;-)
PR #152 is merged so I’ll submit a new PR ASAP. ;-)

MockbaTheBorg, 2022-10-02:
Question is: how would CP/M 3 behave in this case. We should then emulate the behavior as close as we can.

geowar1, 2022-10-02:
IIRC vanilla CP/M 3 requires the “SUBMIT” command. ZCPR3 added the auto-submit support.

geowar1, 2022-10-02:
Normally I don’t specify a file type when typing a command… Are there other types supported (at the beginning of the command prompt) other than .COM?

geowar1, 2022-10-02:
The old (pre-PR #152) behavior always set the file type to .COM.
I think the correct fix would be to only override the file type if it’s blank.

geowar1, 2022-10-02:
Note: That would mean disabling the auto-search (default drive, user zero; drive a, current user; drive A, user 0)… also (if the file type wasn’t blank).

geowar1, 2022-10-02:
Ok, new PR (#155) submitted.

geowar1, 2022-10-02:
Note: I also discovered a bug (I introduced in PR #152) that wouldn’t let the DU: command switch back to drive A:. Fixed(?) it also. Please test! ;-)

MockbaTheBorg, 2022-10-02:
@drawkula should be working as expected now. Please check.

also ... if you could do some regression test we would be grateful.

drawkula, 2022-10-02:
So far the only other thing I found is the args being off by one:
```
C1>type subargs.sub
; 1:$1 2:$2

C1>submit subargs a b c

C1$; 1:A 2:B
C1>subargs a b c

C1$; 1:B 2:C
C1>█
```

MockbaTheBorg, 2022-10-03:
I think the .COM over .SUB part is resolved. Will check the parameters.

MockbaTheBorg, 2022-10-03:
@drawkula can you check now?
```
B0>ed args.sub

NEW FILE
     : *i
    1:  ; 1:$1 2:$2 3:$3
    2:  →
     : *e

RunCPM Version 5.8 (CP/M 2.2 60K)

B0>submit args.sub a b c

RunCPM Version 5.8 (CP/M 2.2 60K)

B0$; 1:A 2:B 3:C
B0>args.sub a b c

RunCPM Version 5.8 (CP/M 2.2 60K)

B0$; 1:A 2:B 3:C

drawkula, 2022-10-03:
```
C1>subargs one two three

C1$; 1:ONE 2:TWO
```
Looks ok.  Thanks!

Things like `c $1.c` in submit files just are needed.  E.g. as compile and run script:
```
C1>type cr.sub
era $1.com
c $1.c
$1

C1>cr hchs

C1$ERA HCHS.COM
C1$C HCHS.C
Hi-Tech Z80 C Compiler (CP/M-80) V3.09-16
Copyright (C) 1984-87 HI-TECH SOFTWARE
Updated from https://github.com/agn453/HI-TECH-Z80-C

C1$HCHS
Hello, CP/M!
Hello, StdIO!

C1>_
```

geowar1, 2022-10-03:
So… can this issue be closed?

closed 2022-10-03 by drawkula