MockbaTheBorg/RunCPM #156: CR vs CR LF problem with Lua?

drawkula, 2022-10-03:
```
CP/M 2.2 Emulator v5.7 by Marcelo Dantas
      Built Oct  3 2022 - 00:05:13
       with Lua scripting support
-----------------------------------------
CCP : INTERNAL v2.6   CCP Address: 0xf400

A0>dir *.lua
A: BDOS     LUA  |  HELLO    LUA  |  LUAINFO  LUA
A0>luainfo
RunCPM System Information v1.0.lua

                                  RunCPM Version 5.7 running on Posix
                                                                     CCP is Internal Version 2.6
                CCP loads at address F400h
                                          BDOS is at address FC00h
                                                                  BIOS is at address FE00h
          64256 bytes available for applications

A0>█
```
I typically don't even use Lua, just wanted to test what else except `.COM` and `.SUB` might be executable without adding the extension...

MockbaTheBorg, 2022-10-03:
I am not observing this. Is this a Linux machine? Terminal?
It is using whatever Lua uses for (CR)LF.
`CP/M 2.2 Emulator v5.7 by Marcelo Dantas
      Built Oct  3 2022 - 08:18:18
       with Lua scripting support
-----------------------------------------
CCP : INTERNAL v2.6   CCP Address: 0xe400

RunCPM Version 5.7 (CP/M 2.2 60K)

A0>dir *.lua
A: BDOS     LUA  |  HELLO    LUA  |  LUAINFO  LUA
A0>luainfo
RunCPM System Information v1.0.lua

RunCPM Version 5.7 running on Windows
CCP is Internal Version 2.6
CCP loads at address E400h
BDOS is at address EC00h
BIOS is at address FE00h
60160 bytes available for applications

A0>`

closed 2022-10-03 by MockbaTheBorg

drawkula, 2022-10-03:
I see that effect on Debian11/AMD64 in Xterm and in text-only consoles.

It probably only affects the systems needing (n)curses.

drawkula, 2022-10-03:
SURPRISE!!!

```
CP/M 2.2 Emulator v5.5 by Marcelo Dantas
      Built Oct  3 2022 - 14:14:12
       with Lua scripting support
-----------------------------------------
CCP : INTERNAL v2.3   CCP Address: 0xe400

RunCPM Version 5.5 (CP/M 2.2 60K)

A0>luainfo
RunCPM System Information v1.0.lua

RunCPM Version 5.5 running on Posix
CCP is Internal Version 2.3
CCP loads at address E400h
BDOS is at address EC00h
BIOS is at address FE00h
60160 bytes available for applications

A0>█
```