MockbaTheBorg/RunCPM #161: Problem with v5.8 under Linux/armbian/Raspbian

guidol70, 2022-10-06:
when compiling v5.8 under Linux (debian like) then I do get the follow warnings:

```
make -f Makefile.posix all
make[1]: Entering directory '/home/guido/RunCPM_git/RunCPM'
gcc -Wall -O3 -fPIC -Wno-unused-variable -c main.c
In file included from main.c:17:
ccp.h: In function ‘_ccp_ext’:
globals.h:162:34: warning: ‘drive’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  162 |  #define _RamWrite(a, v)  RAM[a] = v
      |                                  ^
In file included from main.c:54:
ccp.h:478:11: note: ‘drive’ was declared here
  478 |     uint8 drive, user = 0;
      |           ^~~~~
ccp.h: In function ‘_ccp’:
ccp.h:733:45: warning: ‘sprintf’ may write a terminating nul past the end of the destination [-Wformat-overflow=]
  733 |         sprintf((char *) prompt, "\r\n%c%u%c", 'A' + curDrive, curUser, sFlag ? '$' : '>');
      |                                             ^
ccp.h:733:9: note: ‘sprintf’ output between 6 and 8 bytes into a destination of size 7
  733 |         sprintf((char *) prompt, "\r\n%c%u%c", 'A' + curDrive, curUser, sFlag ? '$' : '>');

```

but stange for me is that - when I exit the compiled RunCPM - I cant see anymore the characters which I type (input) and on the output it seems that there some CR/LF missing - or too much?:

```
/home/guido/RunCPM17:53:44]>
total 148         root@npi-neo2-22 [17:53:46]>
         drwxrwxrwx 3 root root   4096 Oct  6 17:49 A
                                                     -rw-r--r-- 1 root root   2048 Oct  6 17:49 CCP-CCPZ.60K
                                                                                                            -rw-r--r-- 1 root root   2048 Oct  6 17:49 CCP-CCPZ.64K
                                                 -rw-r--r-- 1 root root   2048 Oct  6 17:49 CCP-DR.60K
                                                                                                      -rw-r--r-- 1 root root   2048 Oct  6 17:49 CCP-DR.64K
                                         -rw-r--r-- 1 root root   2048 Oct  6 17:49 CCP-Z80.60K
                                                                                               -rw-r--r-- 1 root root   2048 Oct  6 17:49 CCP-Z80.64K
                                   -rw-r--r-- 1 root root   2048 Oct  6 17:49 CCP-ZCP2.60K
                                                                                          -rw-r--r-- 1 root root   2048 Oct  6 17:49 CCP-ZCP2.64K
                               -rw-r--r-- 1 root root   2304 Oct  6 17:49 CCP-ZCP3.60K
                                                                                      -rw-r--r-- 1 root root   2304 Oct  6 17:49 CCP-ZCP3.64K
                           -rwxr-xr-x 1 root root 104008 Oct  6 17:49 RunCPM
                                                                            root@npi-neo2-22 [17:53:55]>
```

I have changed my prompt from a colorful one to a simple one, but have the same problems.
**I opened a new session. Prompt and input/out does work again**,
 _but_ after starting/exiting RunCPM the input/output is affected the same way.
Also to change the TERM variable (xterm, ansi or linux) doesnt help :(

guidol70, 2022-10-06:
@MockbaTheBorg 
I checked what was changed for Linux (posix) and found only the 21 days old
https://github.com/MockbaTheBorg/RunCPM/commit/aaaa91b055bcad0a05c81c2ec90e0cfe9cee9aaa

where a new initscr was added.
I **commented out** this (void)initscr(); in _console_init **and recompiled.**..
Now after exting RunCPM the console (on X-Window-System) **does work fine** like before using RunCPM v5.8

Seems that the initscr isnt as good for linux as for BSD?

drawkula, 2022-10-06:
I can confirm this for Debian11.

My RunCPM typically runs in an own `Xterm -e ...` incantation which terminates after RunCPM exits, so I previously only noticed Lua misbehaving.

Probably related:
—▷ Issue https://github.com/MockbaTheBorg/RunCPM/issues/156

Ping @dressupgeekout (PR https://github.com/MockbaTheBorg/RunCPM/pull/150)?

guidol70, 2022-10-06:
> Ping @dressupgeekout (PR #150)?

https://github.com/MockbaTheBorg/RunCPM/commit/aaaa91b055bcad0a05c81c2ec90e0cfe9cee9aaa
is [PR #150](https://github.com/MockbaTheBorg/RunCPM/commit/aaaa91b055bcad0a05c81c2ec90e0cfe9cee9aaa)
;) - so I think its the init_scr that cause the problem when exiting RunCPM.
Dont know if it is also the case for the LUA-problem.

dressupgeekout, 2022-10-06:
Whoops. Does it help to add a call to `endwin()` when RunCPM exits? #150 really should have included that >.<

Might also help to replace the `system("clear")` in _clrscr() with the `clear()` function from Curses.

guidol70, 2022-10-07:
> Whoops. Does it help to add a call to `endwin()` when RunCPM exits? #150 really should have included that >.<
> 
> Might also help to replace the `system("clear")` in _clrscr() with the `clear()` function from Curses.

 `endwin()`  does the trick ;)

with
```
void _console_init(void) {
	(void)initscr();
```
and
```
void _console_reset(void) {
	tcsetattr(0, TCSANOW, &_old_term);
	(void)endwin();
}
```
the terminal/prompt does work fine after exiting RunCPM :)

But I couldnt find a working replacement-syntax for clear() :(

https://www.codegrepper.com/code-examples/c/c+ncurses+clear+screen
also does show 
`system(“clear”);`
as ncurses call to clear the screen (which we already use)

MockbaTheBorg, 2022-10-08:
I have pushed a few changes to try and eliminate the warnings above.

Now, I have a question about lncurses:
Is it generically necessary? Or is it for one specific use case?
I am using RunCPM without ncurses built in and I have regular formatting on Wordstar, Spreadsheets, etc. Everything works fine.
What is the use case for ncurses? If it is specific I would prefer to remove it.

guidol70, 2022-10-08:
> I have pushed a few changes to try and eliminate the warnings above.

Did work fine :)

> Now, I have a question about lncurses: Is it generically necessary? Or is it for one specific use case? I am using RunCPM without ncurses built in and I have regular formatting on Wordstar, Spreadsheets, etc. Everything works fine. What is the use case for ncurses?

for me - I dont need it. Have it only because abtraction_posix does include it.
For Windows-TDM it work without it, but I dont tested it on Linux without ncurses.
Normally I would think we donz use any ncurses (now only init_scr and endwin)
Is it used for console programs in Dektop/xterm-consoles?

MockbaTheBorg, 2022-10-09:
I do not think ncurses is required. I have tried different Linuxes and no issues. Both on VMs and Windows WSL.
I will remove it and if anyone needs it for a specific reason then they can fork and enable, as it is not generically required.

closed 2022-10-09 by MockbaTheBorg