MockbaTheBorg/RunCPM #166: ESP32 C3 Support

apcameron, 2022-11-15:
I have managed to get this running on my ESP32 C3 which has a RISC-V Processor.
I added a SD Card and wired it according to the pins listed below
Below is my hardware config file called esp32-c3.h

``#ifndef ESP32_H
#define ESP32_H

// SPI_DRIVER_SELECT must be set to 0 in SdFat/SdFatConfig.h (default is 0)

SdFat SD;
#define SCK         0
#define MOSI        2
#define MISO        1
#define SD_CS       10
#define SPIINIT SCK,MISO,MOSI,SD_CS // sck, miso, mosi, cs 

#define SDMHZ 20
#define SDINIT SD_CS,SD_SCK_MHZ(SDMHZ)
#define LED 5
#define LEDinv 0
#define BOARD "ESP32 C3"
#define board_esp32
#define board_digital_io

uint8 esp32bdos(uint16 dmaaddr) {
	return(0x00);
}

#endif

MockbaTheBorg, 2022-11-15:
Awesome news. Have you joined our Discord already? The invite link is on the readme.

guidol70, 2022-11-15:
@apcameron 
I also did get RunCPM to work on a ESP C3 last May ;)
[image]

Does also work on a ESP S2.

The next  test will be the Lilygo T-Dongle S3 with ESP-S3 MCU ;)

guidol70, 2022-11-16:
This is my 
#include "hardware/esp32/espc3-32s.h"
with comments:

```
#ifndef ESP32_H
#define ESP32_H

SdFat SD;

//ESPC3-32S
//SDCard-Connection-Definition
#define SS    1     // SS/C(ard/hip)S(elect)
#define SCK   7     // Clock
#define MISO  8
#define MOSI  2

// GPIOs 14,15,16,17 are only for the internal SPI-Flash and cant be used :(
// GPIO 18 & 19 is for the white Power-LED
// GPIO 3,4 & 5 are for the R G B LED
// with GPIO 1 & 2 the Power-LED doesnt shine so bright when using SPI

#define SPIINIT SCK,MISO,MOSI,SS
#define SPIINIT_TXT "7,8,2,1"

#define SDINIT SS, SD_SCK_MHZ(SDMHZ)     // ESP32 >= Core v2.0.1

#define SDMHZ 19
#define SDMHZ_TXT "19"
#define LEDY		18 // LED Yellow/Orange
#define LEDW		19 // LED White
#define LEDR		3  // LED Red
#define LEDG		4  // LED Green
#define LEDB		5  // LED Blue
#define LED 		3  // LED used as Disk-LED
#define W_LED 		3  // LED used as write-Disk-LED
#define R_LED 		4  // LED used as read-Disk-LED
#define LEDinv 0
#define BOARD "ESPC3-32S"
#define board_esp32
#define board_digital_io

uint8 esp32bdos(uint16 dmaaddr) {
	return(0x00);
}

#endif
```

apcameron, 2022-11-16:
> Awesome news. Have you joined our Discord already? The invite link is on the readme.

No I prefer not to use discord.

MockbaTheBorg, 2022-11-16:
I understand.   It is just that there we have a team of users always online, so often things are answered quicker.
But hey, happy that you are enjoying RunCPM.