MockbaTheBorg/RunCPM #169: Build errors in cpm.h

teach51, 2022-11-23:
Downloaded the zipped Linux version for MATE Ubuntu onto an old Dell Latitude D830. Compile failed with a bunch of errors. Found and corrected three errors in cpm.h file: "=" missing after DEV_FREE in line 83, F_LOCKFILE truncated to F_LOCK in line 1343, and F_UNLOCKFILE truncated to F_UNLOCK in line 1353. Program compiled and is now running. Don't know much about programming anymore and never used github, but I hope this can save someone else time.

guidol70, 2022-11-23:
@teach51 Thanks - found the same solution and had already created a [Pull-request](https://github.com/MockbaTheBorg/RunCPM/pull/168/commits/0247053e953ba07d0401c4986cde7baa50a6505d) to solve it , which has only to be accepted by @MockbaTheBorg  ;)

MockbaTheBorg, 2022-11-23:
Sorry guys. Been dealing with a country on the brink of a civil war. 😣
Priorities had to shift.
I have merged the pull request. Please check.

teach51, 2022-11-23:
Downloaded zip, recompiled on a different machine and had no problems. Thanks for the quick fix.

MockbaTheBorg, 2022-11-23:
Awesome! Thank you!

closed 2022-11-23 by MockbaTheBorg