MockbaTheBorg/RunCPM #170: Allow RunCPM to be used as a Github Action for SUBMIT jobs.

ravn, 2022-12-17:
I have this ancient hobby project where I disassembled the BIOS on my 8" machine to put in more stuff, and in 2021 got the bits of the old diskettes, and have used RunCPM on my Mac to replicate my toolchain back then.

I now would like to leverage Docker and Github Actions to automate this.  The tricky part was getting a command line into RunCPM but this can be done by generating A/0/$$$.SUB.

ravn, 2022-12-17:
I was expecting I could assign myself, but apparently not :-/  This is a placeholder, not a bug report.

MockbaTheBorg, 2022-12-20:
I have assigned you ... though at this point I believe it is already resolved?

ravn, 2022-12-21:
Getting there.  I want to be able to transparently use this as an action, so more work is needed.

MockbaTheBorg, 2022-12-21:
Cool ... I will close this thread so I don't get scared when I open the main page ... lol.
Then we continua to use it to report success, even closed.
Or the Discord channel.

closed 2022-12-21 by MockbaTheBorg