MockbaTheBorg/RunCPM #171: COPY program

kenmartin-unix, 2023-01-14:
Hi. Is there a COPY.COM program? 

Followed directions and setup and runs OK.

Good Job.

Ken

MockbaTheBorg, 2023-01-14:
Hi Ken,

Copy of files on CP/M is achieved by the PIP.COM program.
Examples: PIP B:XXX.TXT=A:ZZZ.TXT or PIP A:=B:TTT.COM

Cheers,
Marcelo.

kenmartin-unix, 2023-01-14:
Marcelo,

Thanks, that works great!

Also, I have been unsuccessful getting a 'C' compiler working.
Can you point me to one that should work? Link please.

Thanks
Ken

MockbaTheBorg, 2023-01-14:
I have shared my CP/M collection here: 
https://drive.google.com/drive/folders/11WIu8rD_7pIDaET7dqTeA73CvX0jkxz2?usp=sharing

Drive C0: has Hi-Tech C compiler which works.
There are other C compilers there too, among a zillion other things.
Most, if not all, the folders contain an INFO.TXT file describing what is there.

Have fun!

closed 2023-01-14 by MockbaTheBorg

kenmartin-unix, 2023-01-14:
Marcelo,

What are the purpose of the CCP files in the main folder?

You were not kidding zillions of things.

You may cause me to try an port over a BWBASIC interpreter I have for DOS/Linux..

Thanks
Ken

kenmartin-unix, 2023-01-14:
Marcelo,

Working great!!! Good job.

Got a question example drive C2 has a C source if I try and execute A0 TE.com
it says no file. However if I go to A0 I can see TE.com.

How do I transverse (move stuff) between C1 C2 and A0 and A1. Then Execute it?

Sorry for so many questions. The CP/M manual I have does not cover slices.

Appreciate it Ken.