MockbaTheBorg/RunCPM #173: Going Crazt

kenmartin-unix, 2023-01-17:
TheBorg,

I must be going Crazy. I got a few CP/M 'C' programs that just won't compile with any of your 'C' compilers.

One is copy.c, ed.c and stat.c:

[copy.txt](https://github.com/MockbaTheBorg/RunCPM/files/10436474/copy.txt)
[ed.txt](https://github.com/MockbaTheBorg/RunCPM/files/10436475/ed.txt)
[stat.txt](https://github.com/MockbaTheBorg/RunCPM/files/10436481/stat.txt)

The extensions where changed from C to TXT to allow for uploading.

Maybe I should throw in the towel and sit in my rocking chair...

Ken

Xalior, 2023-01-17:
Not all C compilers are created equal.  At-least two of those are written for (at a guess, based on the headers) the Amsterdam Compiler Kit -- which is a cross-compiler, not a CP/M native one.

It's very uncommon to get anything by the most simple or straight forward items of modern code and have them work on a randomly picked hosted CP/M C compiler - most CP/M C compilers were developed long before modern C standards came into practice.

drawkula, 2023-01-17:
`copy` and `stat` build with `ack`.
`ed` yields 150 lines of warnings and errors.
```
C2>copy
Syntax: copy [/uf] [<inputspec...>] <outputspec>
Options:
  F: overwrite output
  U: unbuffered
<inputspec> may contain wildcards. <outputspec> may be just a drive.

C2>█
```
`stat`'s output looks plausible too.
```
(yeti@kumari:12)/home/wrk/runcpm/run/C/2$ cat Makefile 
all: COPY.COM ED.COM STAT.COM

COPY.COM: copy.c
        /opt/ack/bin/ack -mcpm -fp -O6 $< -o $@

ED.COM: ed.c
        /opt/ack/bin/ack -mcpm -O6 $< -o $@

STAT.COM: stat.c
        /opt/ack/bin/ack -mcpm -fp -O6 $< -o $@
(yeti@kumari:12)/home/wrk/runcpm/run/C/2$ █
```

ratboy666, 2023-01-18:
> compile with any of your 'C' compilers.unit

Just peeked at copy.txt

The ONLY CP/M native C compiler that has any hope of this is - HiTech
C. It may need
the addition of definitions of things like  uint8_t, etc. 

Note that The ANSI standard was still 2 years in the future when this
compiler was released!

But, on a quick read, it seems doable (that is, getting it working).

FredW

MockbaTheBorg, 2023-01-20:
I have updated my CPM collection (link on the readme file) with these CPMish tools above. (among other things)
COPY.COM seems to work fine under RUnCPM.
QE.COM doesn't, as I extracted the executables from a KayproII image. The others seem to work fine.

closed 2023-01-20 by MockbaTheBorg