MockbaTheBorg/RunCPM #175: Feature request: Console on ESP32 TTL Serial port

frnno967, 2023-01-24:
I read that by default the console is over the USB connection. I would like to request if it could be made possible for the console IO to take place over the RX/TX connection of the ESP32 instead. This would allow me to build the ESP32 with RunCPM into the cabinet of a Heathkit H89 computer, giving it a nearly original experience without the failing electronics of the vintage CPU board itself. By interfacing the RX/TX lines of the ESP32 to the terminal then it would fully replace the old Z80 computer within. Thanks!

MockbaTheBorg, 2023-01-24:
Hi @frnno967 ,

This sounds like a cool project. 
Are you on our Discord group? Memory may fail me, but I believe someone already did it there.

MockbaTheBorg, 2023-01-24:
Also, for serial access we use the regular Arduino Serial.
This is all RunCPM does:
     #define SERIALSPD 9600
     Serial.begin(SERIALSPD);
     while (!Serial) {	// Wait until serial is connected
Whatever method you use today to access the Rx/TX connection would work here.
You would just need to replace these lines and give it a try.

guidol70, 2023-01-24:
> I read that by default the console is over the USB connection. I would like to request if it could be made possible for the console IO to take place over the RX/TX connection of the ESP32 instead. 

@frnno967  You should check on which serial-port your serial TTL-line is.
Then you have to change the RunCPM.ino like @MockbaTheBorg mentioned
**AND**
in the tab-file "**abstraction_arduino.h**" you have also to replace the serial-Port "**Serial.**"
with the portnumber of your serial TTL-line e.g. "**Serial1.**" or "**Serial2.**"

This can be done by a simple find&replace via the Arduino-IDE menu.

Normally (on a ESP32-board) the "Serial." (which is like Serial0. in counting) 
is the USB-serial-port.

Only on a STM32-Board the USB-serial port was named "**SerialUSB.**"

frnno967, 2023-01-24:
Understood. Thank you for the responses and I'll try it and let you know if any issues are encountered.

drawkula, 2023-01-24:
Related:  https://github.com/MockbaTheBorg/RunCPM/issues/85

MockbaTheBorg, 2023-01-24:
I knew my memory was not too bad.
Unfortunately, since 2019 a lot of things have changed and I can't dedicate as much time to RunCPM as before.

MockbaTheBorg, 2023-01-24:
To prevent me from seeing an issue open here and getting anxious, and as we are already discussing this already on the discord, I will close this one.

closed 2023-01-24 by MockbaTheBorg

drawkula, 2023-01-25:
Code related discussions belong here.
Forcing RunCPM users into discord is not an option.

MockbaTheBorg, 2023-01-26:
Don't see it as "forcing". It is just a matter of convenience to whoever wants answers quicker.
To me actually doesn't matter, but we are already discussing this specific subject in the Discord. So I see no reason to keep it open here.

The only differences between here and Discord are:
. The eMail to which updates to these threads get sent to is not my main eMail, and I don't look at it often.
. I do not open these GitHub pages often either, in fact I very seldom do it.
. At the Discord I look daily, due to other software channels there which I am also developing for, not RunCPM related.

So if someone wishes to discuss a subject on Discord, there will be a much higher chance of dynamic interactive conversation, not only with me but with other folk which are always online there.
Here also works, it is just that here the answers might come 18, 24 or even more hours later.

But seriously, to me doesn't matter. I am fine with either way.