MockbaTheBorg/RunCPM #176: BASIC/Z compiler not supported

riwide, 2023-04-10:
BASIC/Z runs on CP/M 2.2, 3, Turbodos etc but fails on RunCPM v6
E.g. run BZ.COM then attempt LOAD TESTHEX which seeks to load the TESTHEX.BZS file. This will fail on RunCPM  with file not found, but works on CP/M 2.2
Documentation is available for BASIC/Z on Archive.org. I've attached the software in BASICZ.ZIP containing BASICZ.LBR (files within are originally from SEBHC H89emu, but seem generic CP/M)
[BASICZ.zip](https://github.com/MockbaTheBorg/RunCPM/files/11189941/BASICZ.zip)

MockbaTheBorg, 2023-04-13:
@riwide I have submitted a fix that should work fine.
Please give it a try and let me know.

guidol70, 2023-04-14:
I tested the diskh-patch on the Pico and it seems to work, but sometimes I do get a BDOS-Error after working with BASIC/Z on the SDCard which can only be resolved with some resets or completly power off the Pico :(

I hadnt problems with that card/Pico in the past. 
Got this problem if I got BASIC/Z in /A/0 and the SDCard-Init problem is away, when I got the original RunCPM /A/0 Disk in that directory.

BTW: Also the RZ.COM (with RZ IINSTALL) doenst have this "File not found anymore" with the patches disk.h

[image]

riwide, 2023-04-15:
I have tested BZ and used a variety executive commands that interact with CP/M and it seems to be operating well.

closed 2023-04-15 by riwide