MockbaTheBorg/RunCPM #177: NULU open lbr files does not work

riwide, 2023-04-15:
with NULU.COM and my library file prg2304.lbr
nulu prg2304.lbr cannot find the file whereas on real CP/M it opens the lbr file such that -l can be used to list contents
[NULU.CPM.zip](https://github.com/MockbaTheBorg/RunCPM/files/11238996/NULU.CPM.zip)
[prg2304.lbr.zip](https://github.com/MockbaTheBorg/RunCPM/files/11239001/prg2304.lbr.zip)
rename nulu.cpm.zip to nulu.com
rename prg2304.lbr.zip to prg2304.lbr
Nulu normally works correctly on cp/m 2.2 and cp/m 3
I think the version I have supplied is v.152 which is the newest that I am aware of.
I tried with several LBR files, in case one was unintentionally corrupt.
prg2304.lbr contains lbr files for each of my "curated" programming languages and tools.

MockbaTheBorg, 2023-04-15:
That might be related to similar file check as on the BZ.COM.

I will test it tomorrow with the debugger and try to understand why it is
not finding the file.

MockbaTheBorg, 2023-04-15:
I just got home and gave it a try.
I was unable to replicate.
Copied and renamed the files, loaded nulu, opened the library, listed and extracted.
All works fine.
Are you having problem on Linux? Windows? Devices?

riwide, 2023-04-16:
I was on Linux, using latest build 6 with basic fix, then backed off to 5.7

With 6, make posix clean, then make posix build
I tried several different lbr files.

I will try on Windows,  and if it works take exact nulu and lbr across to
Linux and try again.

I did note some strangeness under runcpm on Linux with drive P: when the
underlying  Linux folder had a mix of uppercase filenames and lowercase
filenames. This was noted after I couldn't run nulu which would have
generated u/c filenames, the lbrate and mlbr extracted under Linux with l/c
names. Need to look at that more closely.

I use Linux Mint 21.1

I have done more with Basic/z and that still works. Load, save, compile,
exec.

Hopefully I can respond more later today.

Cheers
Richard

riwide, 2023-04-16:
OK, think I have it narrowed down.

On windows - name the lbr file in lowercase e.g prg2304.lbr, then runcpm
dir and nulu will display the file as uppercase, and nulu can work with and
extract that file - i.e. all works

On linux - name the lbr file in lowercase e.g prg2304.lbr, then runcpm dir
and nulu will display the file as uppercase, *but nulu cannot open that
file - i.e. broken.*

I suppose that Linux's use of both upper and lower case names is confusing
some aspects of runcpm.

I am not sure whether you consider this as a bug or a feature. I probably
hit the problem by giving a lowercase name in minicom for save of  xmodem
transfer to linux of the lbr file.

Cheers
Richard

MockbaTheBorg, 2023-04-16:
On Linux we kinda created a standard of naming everything upper case.
I do have everything uppercase in windows too, just for fun, as it doesn't
matter.
On Linux it might create issues.
In theory, when in Linux, files with "invalid" characters (which includes
lower case) are like if they're not there.
(Same for folder and CCP names)
So it is a "limitation" not a bug.

riwide, 2023-04-16:
Thanks and apologies to you for any time wasted.

I've got my rename command installed under Linux  and learnt how to do
"find . -depth | xargs -n 1 rename 's/(.*)\/([^\/]*)/$1\/\U$2/' {} \;" in
case I get a collection of lower case files again.

I used to use Windows but around Christmas too many Windows 11 blue screens
made me switch to Linux. I have to keep Windows around for a few
Windows-only programs though.

Regards
Richard

MockbaTheBorg, 2023-04-16:
Zero time wasted, no worries.
I am just sticking to windows 10 now. Windows 11 gives me creeps. If I
wanted it for the looks only I would then get a mac.
I have a feeling windows 11 will be "skipped" like windows 8 (or windows
Vista) was.

wrljet, 2023-04-16:
Other than for Hercules build testing, I'm skipping Windows 11.  :)

MockbaTheBorg, 2023-04-18:
I was using this script:
```bash
#!/bin/bash
shopt -s globstar

for file in **/*
do
  if [[ -f "$file" ]] || [[ -d "$file" ]]; then
    if [[ "${file^^}" != *"RENAME.SH"* ]] && [[ "${file^^}" != *"RUNCPM"* ]]; then
      dest="${file^^}"
      if [[ "$file" != "$dest" ]]; then
        echo "$file" '=>' "$dest"
        mv "$file" %TEMP%
        mv %TEMP% "$dest"
      fi
    fi
  fi
done
```
As it should work both on native Linux and WSL.

I will close this one as it is basically a "limitation" of RunCPM while running on Linux or similars.

closed 2023-04-18 by MockbaTheBorg