MockbaTheBorg/RunCPM #181: ESP32-Wrover kit freeze

Isysxp, 2023-08-22:
This is probably a specific hardware problem.
Env: ESP32 Wrover kit with onboard LCD using  fabgl::ILI9341Controller DisplayController.
CP/M will boot to A> prompt but most apps freeze (eg DIR). This was corrected using a trivial patch:
globals.h	                      static uint8 *RAM;
RunCPM.ino:
    Following:
        _puts(BOARD);
	_puts("\r\n");

	// Allocate MEMSIZE bytes of memory for RAM
	RAM = (uint8*)malloc((size_t)MEMSIZE);
	if (!RAM) {
		_puts("Mem alloc fail.");
		while (1);
	}
Other than this .. a really cool set of apps.

guidol70, 2023-08-22:
@Isysxp  Dear Ian,
Your
`static uint8 *RAM;`
in **globals.h**
seems to conflict (when **RAM_FAST enabled**) either with
`static uint8 RAM[MEMSIZE];`  ( conflicting declaration 'uint8 RAM [65536]' )
in **globals.h**

or  (when **RAM_FAST disabled**)
`static uint8 RAM[MEMSIZE];`
in **ram.h**

When I do disable the old lines in globals.h and ram.h I do get the malloc-error
"Mem alloc fail." from the .ino :(

Kind regards
Guido

Isysxp, 2023-08-22:
Dear Guido,

 Should have seen this one! The malloc patch should be guarded with
#ifdef RAM_FAST
#endif
or change line 7 in ram.h (as per your note) as well to:
static uint8 *RAM;

BW, Ian.

guidol70, 2023-08-23:
> Should have seen this one! The malloc patch should be guarded with #ifdef RAM_FAST #endif or change line 7 in ram.h (as per your note) as well to: static uint8 *RAM;

Dear Ian,
after changing "my code" I had the "Mem alloc fail." - BUT only for one reason :)
For your MemAlloc I had to **enable PS-RAM** in the flashing settings.

My former version didnt used that so I didnt noticed that I had to enable it by your error-message.

So now it does work - **also with VGA16Controller again** :)

Kind regards
Guido

[image]

[RunCPM_VGA32_v6_1_IanMem_23082023.zip](https://github.com/MockbaTheBorg/RunCPM/files/12418212/RunCPM_VGA32_v6_1_IanMem_23082023.zip)

Isysxp, 2023-10-18:
I think so. Guido, are you happy with the system as it stands?

MockbaTheBorg, 2023-11-22:
@guidol70 , if all set we can close this issue.

Isysxp, 2023-11-24:
I agree.

closed 2023-11-24 by Isysxp