MockbaTheBorg/RunCPM #182: Request: Simple variation on LED.BAS as a Hello World using BDOS 9

snakebyte69, 2023-09-07:
100 REM ;### BDOS procedures in machine code ###
110 REM
120 REM ;Compiles with pasmo, rasm, sjasmplus, spasm-ng, 
130 REM ;not with z80asm, not z88dk-z80asm, zasm, zmac
140 REM 
150 REM .Z80
160 REM 
170 REM WRITESTR:   EQU 9h
180 REM BDOS:       EQU 5h
190 REM 
200 REM             ORG 100h
210 REM             LD C,WRITESTR ;110D01
220 REM             LD DE,HELLO ;0E09
230 REM             CALL BDOS ;CD0500
240 REM             RST 0
250 REM 
260 REM HELLO:
270 REM             DB "Hello World!$"
280 REM    
290 REM             END ;z88dk-z80asm won't compile comment out
300 REM
310 REM **** Decimal Equivalent of Hex Opcodes ****
320 REM  0e,09,11,09,01,cd,05,00,c7,48,65,6c6c,6f,20,57,6f,72,6c,64,21,24 
320 DATA 14,9,17,9,1,205,5,0,205,72,101,108,108,111,32,87,111,114,108,100,33,36,-1
330 REM
340 REM ### Load machine code procedures to memory ###
350 REM
360 BD$="" : READ A : WHILE A>-1 : BD$=BD$+CHR$(A) : READ A : WEND
370 ADR=PEEK(VARPTR(BD$)+1)+PEEK(VARPTR(BD$)+2)*256
380 REM
390 REM ### USRx = BDOS CALL 22x ###
400 REM
410 DEF USR0=ADR : DEF USR1=ADR+14 : DEF USR2=ADR+37 : DEF USR3=ADR+51
420 REM
430 REM ### Input PIN number and its value (0 = LOW, 1 = HIGH) ###
440 POKE ADR+74,P : POKE ADR+75,V
450 X=USR0(PEEK(ADR+74)*256+1)
460 X=USR2(PEEK(ADR+74)*256+PEEK(ADR+75))

snakebyte69, 2023-09-07:
Got it to spit junk on the screen and crash

MockbaTheBorg, 2023-10-18:
Hi there @snakebyte69 ,

Sorry to be so late in replying.
Any progress here? I am not sure what the problem is.
The program seems to load a small assembly in memory, to print "Hello World" so it is not printing anything?

Thanks,
Marcelo.

snakebyte69, 2023-10-28:
I was experimenting with inline assembly using BBC Basic and Microsoft Basic.
I got it working in both it doesn't use this method I created a new one.
I worked out the correct calls for BDOS and BIOS from Basic too.

closed 2023-11-22 by MockbaTheBorg