MockbaTheBorg/RunCPM #185: [Q] Difference for RunCPM between TYPE and TYPE in a .SUB?

guidol70, 2023-11-22:
A short question:
I know CLS is a internal command in RunCPM, but additionally I did try to implement 
it again in a short way to use it universally in different CP/M-versions.

I did create a binary file CLS.SEQ (for ANSI) with: Esc[H Esc[2J
also (POS1 and Clear Screen) in HEX:
`1B 5B 48 1B 5B 32 4A`

This does work fine with TYPE CLS.SEQ , but when I create a **SCLS.SUB** 
(because CLS is the internal command) with the following content
`TYPE CLS.SEQ`
it doenst do the same in RunCPM - I do get (if I use SCLS or SCLS.SUB):

```
A0>SCLS

RunCPM [v6.1] => CCP:[INTERNAL v3.0] TPA:[64K]

A0$
?

A0>
```

In the [CP/M-68K of simcpm](https://github.com/linuxha/cpmsim) the SCLS.SUB does work.
Has there to be anything else in a .SUB for RunCPM?

Many Thanks for the information in advance ;)
Kind regards
Guido

[CLS_SEQ.zip](https://github.com/MockbaTheBorg/RunCPM/files/13443772/CLS_SEQ.zip)

MockbaTheBorg, 2023-11-22:
This could be a bug ... I will have to check it.
The behavior should be the same as on regular CP/M.
Is this happening only on internal CCP? Or also on the regular DR CCP (or others)?

MockbaTheBorg, 2023-11-23:
I am not absolutely sure if the problem is with RunCPM.
Here's the test I did:
. Go to A:
. Create a TST1.SUB file with a single line 'type info.txt'.
. Create a TST2.SUB file with a single line 'type info.txt' plus an empty line below it.
When you do submit TST1, the $$$.sub file created contains a correct single line, 128 chars long with 'type info.txt' preceded by the string size and terminated by 00. Which is the correct SUBMIT format.
When you do submit TST2, the $$$.SUB file is created empty. Both with SUBMIT.COM and SUBMITD.COM, no difference.
Somehow submit seems to be broken, stopping processing when it finds an empty line.
The same exact behavior happens on Z80EMU, so I think RunCPM is doing what it is supposed to do. Crashing.
There's some reference to the problem in this page:
https://forum.vcfed.org/index.php?threads/xsub.56082/

So yes, it is broken, but there's nothing to fix. :)
Closing the issue.

MockbaTheBorg, 2023-11-23:
PS: Grab SUPERSUB.COM from my B0: disk, it is working fine

closed 2023-11-23 by MockbaTheBorg

guidol70, 2023-11-23:
copied SUPERSUB.COM from the B/0 Archive as SUBMIT.COM to A/0 and SCLS.SUB does work fine as you wrote :)

Many thanks for the information ;)