MockbaTheBorg/RunCPM #187: [Q] How about updateing to LUA v5.4.6?

guidol70, 2023-12-30:
Hi,
actuallly in /RunCPM/lua directory is LUA v5.4.4 - Release from the 26. Jan 2022
and at https://www.lua.org/ftp/lua-5.4.6.tar.gz is v5.4.6 - Release from the 14. May 2023

I did exchange the content with the src-dirctory of the new version and hadnt any problems compiling and runnning with the new version.

[image]

Maybe you want to update it? ;)

MockbaTheBorg, 2023-12-30:
I actually intend to pull it and make it a "vendor" thing. So when someone
comes, they get the latest Lua.
Though I have no idea how to do it.
I will have to figure it out.

MockbaTheBorg, 2024-03-10:
Lua is now a submodule, which should be populated with the latest version when RunCPM is cloned.

closed 2024-03-10 by MockbaTheBorg