MockbaTheBorg/RunCPM #189: Command switch to select base virtual-disk directory

Kroc, 2024-02-19:
I'm using RunCPM to test a Z80 assembler I'm writing in Z80 (https://github.com/Kroc/v80) with the goal to rewrite the assembler in itself and do the assembling from within RunCPM. Having the virtual disk default to the binary location is somewhat awkward as it means copying files into it as part of the build script and it lacks the flexibility to use different virtual disks without also making multiple copies of the binary! Could we please have a command-line switch to select which directory will serve as the virtual disk root.

MockbaTheBorg, 2024-03-10:
Hi there ... to be honest I do not have plans to do it, because a command line parameter is a Windows/Linux only thing.
I intend to keep the code in a way that it will be as equal as possible for Windows, Linux, Mac, Arduino, Teensy, ESP32, etc. And on the microcontrollers a command line switch is not implemented.
Not to mention some other chips and FPGAs where I have it unreleased.
I encourage people to clone the repository and steer the code towards their specific usage. Some people have added VGA support, etc.
So please feel free to clone and make your changes.

closed 2024-03-10 by MockbaTheBorg

0cjs, 2024-04-11:
I've got a couple of comments about this.

First, there's already a large difference for just this thing between the microcontroller and  Windows/POSIX versions: the microcontrollers always use the root of the filesystem, which is quite reasonable for them, but obviously isn't for the Windows/POSIX systems (or "PC" systems, as I'll call them from now on).

Thus, the PC systems need to choose something else. That choice is currently the directory from which the program is run, but that's a bit odd for PC systems: it's not normal that you'd put data directories (particularly multiple ones, such as `A/`, `B/`, etc. under `/usr/local/bin/` or `C:\Program Files\`, which are directories that a non-root user can't even write. (Of course we get around this, at the moment, by copying the binary all over the place, or maybe symlinking on POSIX, but having lots of copies of a given program isn't particularly desirable for various reasons.)

A better choice might have been to use the current working directory, since that is entirely under control of the user running the program, but at this point I think it's not unreasonable to say that that bird has flown, unless people are a lot more ok with a backwards-incompatibility than I imagine. (I'm ok with it myself, but that's for reasons that don't apply to a lot of people.)

Given that, the only way to improve the situation on PCs seems to be to add a command-line option. This isn't as bad as you might be thinking, for two reasons. First, it could be done in just `main.c`, which is PC-only; microcontrollers use a completely separate `RunCPM.ino` instead, which doesn't even have a `main()` function. So adding a command-line parameter at the code level is really just making a change to an area where the code is already completely split, and where we're already running PC code that doesn't exist on the microcontrollers (the `_host_init(argc, argv)` function and its contents, which is the only thing that needs to change).

I also feel that the "keep PC and microcontroller the same" thing isn't really followed anyway, since there's a significant amount of code (much larger than what's in this proposal) in the microcontroller version for doing things like manipulating GPIO pins. If that much platform-specific code is ok for microcontrollers, it seems to me that a much smaller bit of platform-specific code should be ok for PC.

All that said, it might also be the case that one wants to make available alternate directories even on microcontroller systems. Imagine that someone's doing a more general "emulate a bunch of things" microcontroller system; they would have a top-level menu program to let you select the emulator and game, and that would probably want to be able to parametrise the directories used by the microcontroller version. 

lights