MockbaTheBorg/RunCPM #191: Z80ASM (and M80) hang

rtrussell, 2024-05-03:
RunCPM (built on Windows 11 using mingw) seems to run OK, but if I attempt to assemble a short program using either M80.COM or Z80ASM.COM it hangs:

```
RunCPM Version 6.2 (CP/M 60K)
B0>z80asm sorry.bbb/r
Z80ASM Copyright (C) 1983-86 by SLR Systems Rel. 1.32 #RunCPM
 SORRY.BBB/R
```
I have to type Ctrl+C at this point, whereupon it exits RunCPM back to the Windows command prompt.

This is the code I'm trying to assemble:

```
	GLOBAL	CLG
	GLOBAL	COLOUR
	GLOBAL	DRAW
	GLOBAL	ENVEL
	GLOBAL	GCOL
	GLOBAL	MODE
	GLOBAL	MOVE
	GLOBAL	PLOT
	GLOBAL	SOUND
	GLOBAL	ADVAL
	GLOBAL	POINT
	GLOBAL	GETIMS
	GLOBAL	PUTIMS
;
	EXTRN	EXTERR
;
CLG:
COLOUR:
DRAW:
ENVEL:
GCOL:
MODE:
MOVE:
PLOT:
SOUND:
ADVAL:
POINT:
GETIMS:
PUTIMS:
	XOR	A
	CALL	EXTERR
	DEFM	'Sorry'
	DEFB	0
;
	END

```

rtrussell, 2024-05-03:
> RunCPM (built on Windows 11 using mingw) seems to run OK, but if I attempt to assemble a short program using either M80.COM or Z80ASM.COM it hangs:

On further investigation, the problem appears to be that M80 and Z80ASM are attempting to create an output file, but the file **already exists** and is set to **read-only**!  Rather than issuing an error message or failing gracefully, RunCPM is hanging in those circumstances.

MockbaTheBorg, 2024-05-03:
Interesting. Question:
If you have a read-only file in B:, and then you try to PIP something into it, like "pip B:FILE.RO=A:1STREAD.ME", does it also hang?
Or it only happens when using M80/Z80ASM?

rtrussell, 2024-05-03:
> If you have a read-only file in B:, and then you try to PIP something into it, like "pip B:FILE.RO=A:1STREAD.ME", does it also hang? Or it only happens when using M80/Z80ASM?

PIP hangs too.  Again the only way out is Ctrl+C, and that exits RunCPM.

MockbaTheBorg, 2024-05-03:
Should be fixed now ... it went in a tight loop.
Please verify.

closed 2024-05-03 by MockbaTheBorg