MockbaTheBorg/RunCPM #195: Unable to build latest main branch

TurBoss, 2024-05-15:
Hello,

I was trying to get the esp32 and teensy41 to build but getting this error

```
In file included from /home/turboss/Arduino/RunCPM/RunCPM/RunCPM.ino:7:
/home/turboss/Arduino/libraries/SdFat/src/SdFat.h:462:2: warning: #warning File not defined because __has_include(FS.h) [-Wcpp]
  462 | #warning File not defined because __has_include(FS.h)
      |  ^~~~~~~
In file included from /home/turboss/Arduino/RunCPM/RunCPM/RunCPM.ino:41:
/home/turboss/Arduino/RunCPM/RunCPM/ccp.h: In function 'void _ccp()':
/home/turboss/Arduino/RunCPM/RunCPM/ccp.h:698:38: error: '_sys_fopen_r' was not declared in this scope; did you mean '_sys_fopen_w'?
  698 |                         FILE* file = _sys_fopen_r((uint8*)AUTOEXEC);
      |                                      ^~~~~~~~~~~~
      |                                      _sys_fopen_w
/home/turboss/Arduino/RunCPM/RunCPM/ccp.h:699:39: error: '_sys_fread' was not declared in this scope; did you mean '_sys_readseq'?
  699 |                         blen = (uint8)_sys_fread(&dmabuf[0], 1, 128, file);
      |                                       ^~~~~~~~~~
      |                                       _sys_readseq
/home/turboss/Arduino/RunCPM/RunCPM/ccp.h:712:37: error: invalid initialization of reference of type 'File&' {aka 'File32&'} from expression of type 'FILE*'
  712 |                         _sys_fclose(file);
      |                                     ^~~~
In file included from /home/turboss/Arduino/RunCPM/RunCPM/RunCPM.ino:17:
/home/turboss/Arduino/RunCPM/RunCPM/abstraction_arduino.h:76:24: note: in passing argument 1 of 'void _sys_fclose(File&)'
   76 | void _sys_fclose(File& f) {
      |                  ~~~~~~^
```

thanks

MockbaTheBorg, 2024-05-15:
I remember there was some change needed to the sdfat config file for these
devices.
I am away from the computer but it should be on the readme file on GitHub.
Please check and let me know.
If that doesn't work we move to the next step.

Thanks!

TurBoss, 2024-05-15:
Hello, for the teensy is not needed but yes for the esp32
no luck so far

i tried to figure when started breaking and found 7be035defb120072dbdbae4f6c940a57dbe4e6dc this commit

since this I'm unable to compile using arduino

Thanks

MockbaTheBorg, 2024-05-16:
Give it a try now ... I have reworked the autoexec functionality, and it should build correctly and work for the Arduinos now.

TurBoss, 2024-05-17:
Hello,

It builds ! :partying_face: 

Thank you

closed 2024-05-17 by TurBoss