MockbaTheBorg/RunCPM #196: 16-bit DOS as a build target ?

Arawn-Davies, 2024-05-23:
Hi, I've really enjoyed the simplicity of running CPM binaries that RunCPM has provided, so I've used it on both desktop and on an Arduino Due project of mine.

I've recently bought a Book 8088 V2 that has a NEC V20 CPU, while 8-bit external bus it's of course a 8086 internally with improvements etc. It's also got a full 640Kb (more than anyone would ever need surely?) 😂 and an XT-IDE ROM that provides modern compact flash support. 

I feel like it could be a great platform to run RunCPM on, as it supports MS-DOS 6.22 but I don't know how much of RunCPM could be used with a suitable compiler :) I'm happy to look into it or see if there's any other solutions, as the BIOS on the Book 8088 V2 makes it hard to use CPM on it at all

SEGStriker, 2024-05-23:
> Hi, I've really enjoyed the simplicity of running CPM binaries that RunCPM has provided, so I've used it on both desktop and on an Arduino Due project of mine.
>
> I've recently bought a Book 8088 V2 [...]

Not at all, actually V20 has 8080 hardware emulation mode, there is at least one emulator, which is using that. Read https://en.wikipedia.org/wiki/NEC_V20 (and other sources) to learn more about it.

MockbaTheBorg, 2024-05-23:
Hi @Arawn-Davies , I do not support DOS anymore, that support has been dropped a while ago.
If you are part of our Discord group, feel free to ask there. I know that some folk there are still building it on DOS.
I had to simplify my dedication to the project as "real life" got in the way.

Arawn-Davies, 2024-05-28:
> Not at all, actually V20 has 8080 hardware emulation mode, there is at least one emulator, which is using that. Read https://en.wikipedia.org/wiki/NEC_V20 (and other sources) to learn more about it.

I've recently dug my 486 out of the attic which won't have built-in 8080 emulation, I get there are plenty of emulators out there but having something like RunCPM where I don't have to worry about diskdefs etc. would be great :) I'll give it a try with a 486 linux distro if that works, or try ELKS as a target 😂

Arawn-Davies, 2024-05-28:
> Hi @Arawn-Davies , I do not support DOS anymore, that support has been dropped a while ago. If you are part of our Discord group, feel free to ask there. I know that some folk there are still building it on DOS. I had to simplify my dedication to the project as "real life" got in the way.

no worries  I'll check discord out, thank you for your and others hard work :)

ravn, 2024-06-03:
Very good CP/M emulators exist for plain DOS - I would suggest you use one of those if you actually need it for real work. If this is just for fun and you have time to spare, getting RunCPM to run under plain DOS is a great coding exercise

guidol70, 2024-06-03:
@Arawn-Davies 

RunCPM may be for your  XT/V20 a little too big.
The only DOS-Port I know ;) is my port, but it does need a 486 and up, because it does use a 32Bt DPMI-Extender.
Downloadable at https://github.com/guidol70/RunCPM_Windows/raw/main/GL20240529_RunCPM_DOS_v6_3_Starterpack.zip

[image]

Directly for your Book V2 with NEC V20 CPU I would recommend you the following CP/M-Emulator, because they recognize you V20-CPZ and do use the Hardware-8080 Emulation inside your V20-CPU.
The are named V2080 / ZRUN or 22NICE
With these you can run a CP/M-program like MBASIC.COM with the following DOS-Commandline:
V2080 MBASIC.COM
ZRUN MBASIC.COM
22NICE MBASIC.COM

As information:
`
An unusual feature of the NEC V20 was that it added an Intel 8080 emulation mode, in which it could execute programs written for the Intel 8080 processors. The instructions BRKEM executed in 8086 mode (NEC used a different notation for the instructions than Intel and BRK in NEC notation = INT in Intel notation) and RETEM and CALLN executed in 8080 mode was used to switch or return to or from the emulation mode. There were some programs which allowed to run 8080-based CP/M-80 programs on a MS-DOS machine: V2080 CPMulator (later ZRUN) by Michael Day and 22nice from SYDEX .
`

You can find these emulator at
http://cd.textfiles.com/230/EMULATOR/DIVERSE/CPM/

The second way I can recommend ;) is to "cheat a bit" for a very fast RunCPM on your book - while using the Book NEC V20 inly as terminal to a RunCPM on a Raspberry Pico using real RS-232 (not USB-serial)

[image]

If you really want to use this I can build you a special flash-binary for the Pico, which uses GPIO0/1 as serial (from there you need a TTL to RS-232 serial adapter) and a "XT-Baudrate" like 9600 or 19.200 Baud (because most XTs cant do 115.200Baud becaus eof the CPU/serial FIFO
I dont know if the Book has a better buffered serial port.

MockbaTheBorg, 2024-06-17:
Hi folks, I will close this one as it is a non-issue.
Thanks for trying RunCPM in different platforms!

closed 2024-06-17 by MockbaTheBorg