MockbaTheBorg/RunCPM #204: RunCPM can't execute COM file that's definitely there -- filename case issue?

Kroc, 2025-02-08:
The filenames are lowercase on the filesystem, but whilst RunCPM cannot list the files, it can't execute them regardless of case typed!

[image]

MockbaTheBorg, 2025-02-09:
I see it  happens on Linux.
I wonder ... if we have 'xpto.COM', 'XPTO.com' and 'xpto.com' in the same folder, which are perfectly fine Linux names, which one should RunCPM execute when you type XPTO?
The answer is: none! because none of them are XPTO.COM.
Specifically for Linux I had added the following to our readme:
`All the letters for folders/subfolders and file names should be kept in uppercase, to avoid any issues of case-sensitive filesystems compatibility. 
`
So in this case I suggest just use uppercase across the board.

Kroc, 2025-02-09:
Okay, that's fair but I wish I had a choice of uppercase vs. lowercase. Perhaps if a file doesn't exist in the all-uppercase form, check for the all-lowercase form?

SEGStriker, 2025-02-09:
What if the subdirectory has both upper and lower characters, for example ProgAll? So, no way for such a smart check.

Kroc, 2025-02-09:
It wouldn't have to handle all possibilities -- just ALL CAPS and all lower as the two predominate styles.

SEGStriker, 2025-02-09:
I would say that one of predominate styles is to have capital letter at the beginning. Which means it depends on the logic.

MockbaTheBorg, 2025-02-10:
I will actually see if I can make it use the first one. Doesn't matter caps combination.
Then it becomes up to us to prevent duplicates.
But it would ignore case.
I think this would be the best behavior.

feilipu, 2025-02-10:
Just to point out the obvious, the repo is RunCPM. Allowing lowercase would make it RunSomethingLikeCPMish.

CPM expects all uppercase filenames, and many original utilities rely on this. So imho, keep to the standard of CPM 2.2 or later. 2 cents.

MockbaTheBorg, 2025-02-10:
> Nothing changes within the CP/M environment, this would be entirely a
> convenience on the host side

Yes. That would be just to fix (or standardize) on the host side. The cp/m
will continue to be uppercase.

SEGStriker, 2025-02-10:
On the host side it is fully OS dependent - DOS/Windows doesn't really care about the case (Windows can use both upper and lowercase in file names, but still doesn't really care). Simple example:

C:\>dir f*
 Volume in drive C is OS

 Directory of C:\

01/22/2025  04:45 PM    <DIR>          Files
               0 File(s)              0 bytes
               1 Dir(s)  229,678,669,824 bytes free

C:\>dir F*
 Volume in drive C is OS

 Directory of C:\

01/22/2025  04:45 PM    <DIR>          Files
               0 File(s)              0 bytes
               1 Dir(s)  229,678,620,672 bytes free

As shown, it will find "Files" in both cases, even though it contains one capital letter and the rest are lowercase. Same in more complex situations.
But in Linux it is completely the opposite - it really cares about upper/lowercase. So in theory we could have /Files, /files and /FILES in  one directory:

[root@server]# mkdir files
[root@server]# mkdir Files
[root@server]# mkdir FILES
[root@server]# ls -la
total 5
drwxr-xr-x   6 root root   99 Feb 10 01:39 .
dr-xr-x---. 17 root root 4096 Dec 12 20:09 ..
drwxr-xr-x   2 root root    6 Feb 10 01:39 files
drwxr-xr-x   2 root root    6 Feb 10 01:39 Files
drwxr-xr-x   2 root root    6 Feb 10 01:39 FILES

This is a real test, not just manipulation :)
So, how will RunCPM distinguish these 3 directories if all of them exist and are converted by the emulator to uppercase?

MockbaTheBorg, 2025-02-10:
RunCPM ignores directories, it only pays attention to files. And in case of
the files it will be the first found. (That is my plan at least)
And it will ignore case. Leaving up to us to prevent duplicates.
(Unless Linux glob calls prevent me from doing it)

MockbaTheBorg, 2025-02-10:
> Yeah, it is same with files. Directories are nothing else but special kind
> of files, you know that.
> OK, whatever you decide.

What I meant is. The only directories that matter are the "drive" ones. Any
directories inside those are ignored (it is a bug if they aren't).
So if you create a XPTO.COM folder, it will be like if it is not there.

MockbaTheBorg, 2025-02-10:
Ok ... upon revisiting the disk code after all these years the conclusion is:
            can´t be done.
RunCPM treats all filenames internally as uppercase, as it should be.
So, even though I am using the glob() function to collect the directory on Linux in a case insensitive manner today (which means a:v65.com appears as A:V65.COM when you DIR), when you try to run that program it will ultimately be converted to a fopen() request to ./A/0/V65.COM, which doesn´t exist.
To make that exist it would require RunCPM to either work with the names internally everywhere without converting them to uppercase or keep a translation table somehow (which is a different can of worms, as there might be multiple instances of the "same" name with different case combinations. 
It is a lot of code to modify for what is actually a documented "limitation" specific to Linux (remember, RunCPM was born on Windows, Linux is already a port of it).
So at this point, for the specific Linux use case I can only offer one proposition:
Have your CPM filesystem created in an image or USB device formatted with vfat, cd to there and run RunCPM. It will be case insensitive now and everything will work.
I actually use the USB solution, so I can move my CP/M collection between my Windows desktop, my Linux laptop and bring it with me to work when I want to procrastinate.

MockbaTheBorg, 2025-02-10:
Tried to get rid of all the 'toupper' internally to see what would happen. But there's a problem hiding there.
When we type v65 I need to append the .COM extension to make it be looked up and run. That is done by the CCP, not by RunCPM, so we're out of luck there as well.

closed 2025-02-10 by MockbaTheBorg

Xalior, 2025-02-10:
ext4 has support for marking specific directories as case insensitive, which also has options as a workaround.

MockbaTheBorg, 2025-02-10:
That's a good option, I didn´t know about that ... but looks like the ext4 filesystem must have been created with the casefold option enabled to accept it. On my laptop, for example, even though the casefold option seems to be supported by the kernel, the filesystem won't accept it. So I am out of look. You mileage may vary on this one.

Xalior, 2025-02-10:
I suffered the same issue last year -
https://unix.stackexchange.com/questions/747437/how-to-change-the-casefold-ext4-filesystem-option-of-the-root-partition-if-i-on
sorted me out. x

MockbaTheBorg, 2025-02-10:
So, turns out that enabling casefold on my Linux laptop's filesystem, so I could make my CPM directory case insensitive, solved two problems:

1. Files and executables can be named lowercase, uppercase, comination, doesn't matter.
2. You cannot have multiple files with the same name in the folder, because the folder now is case insensitive. (behaves just like Windows folder)

This is then the ideal solution for Linux it seems. Requires a kernel higher than 5.2 though. I would recommend way higher. Mine is  6.11.

Here is how I made it work:

1. Logoff from the laptop completely so I can log in back as root and dismount /home. Needs to be dismounted.
2. Log back as root and enable casefold onto the /home filesystem device:
        umount /home
        tune2fs -O casefold /dev/sdb3    (my /home is at /dev/sdb3, your's may/will be not)
        reboot
3. Create an empty CPM folder inside my /home/mockba.   (you can create somewhere else)
4. Enable casefolding on that folder (has to be done on an empty folder):
        chattr +F CPM
        reboot   (somehow it only worked well after I rebooted again)
5. Copy all the folders and files from my old CPM folder recursively onto this one.
6. Great success!

Now it works under Linux precisely the same as it would on a FAT filesystem.

Kroc, 2025-02-10:
Changing the folder's case-folding is not something that's going to transfer across Git if others download / clone the repo, so I'll switch to using all uppercase for files. Thanks for the consideration!