MockbaTheBorg/RunCPM #206: SearchFirst/SearchNext not returning user number

jgharston, 2025-07-05:
BDOS SearchFirst and SearchNext with drive specified as '?' (such as using DIR ?:) correctly returns all file entries of all users, but appears to not actually return the user number in the returned FCB. For example, with a disk with
A0: FRED.TXT
A1: JIM.TXT
A2: SHEILA.TXT
repeated calls to SearchFirst/SearchNext return:
&00,"FRED    TXT"
&00,"JIM     TXT"
&00,"SHEILA   TXT"

ie, the three file entries, but FCB+0 returning "user 0" for all of them.

[image]

jgharston, 2025-07-05:
This is what is supposed to happen:

[image]

MockbaTheBorg, 2025-09-04:
I ran a quick test today on Linux and could not replicate the issue (under abstraction_posix.h) ... the output was:
```
B>test
CP/M Directory Listing Demo
0:TEST    .COM
0:TEST    .Z80
1:HELLO   .COM
1:HELLO   .Z80

B>
```
So I need more details:
1. which host are you running on?
2. which CCP are you running on?
3. are you using some CP/M application to replace DIR?

Lets fix it.

I have attached the source code for the test I ran. Just save as TEST.Z80 and compile with the included Z80ASM:
```
B>A:Z80ASM TEST
```

[TEST.Z80.TXT](https://github.com/user-attachments/files/22138880/TEST.Z80.TXT)